tinygo-org/bluetooth #9: docs: how to flash board with UF2 bootloader?

deadprogram, 2020-08-29:
How exactly would one go about flashing a board such as the Circuit Playground Bluefruit with the needed code to run a TinyGo Bluetooth application?

deadprogram, 2020-08-29:
I see that the prebuilt binaries for bootloaders that Adafruit has built are all for for v6.1.1 of SoftDevice:

https://github.com/adafruit/Adafruit_nRF52_Bootloader/releases

I assume that probably means that this is the firmware already on the the shipping boards? In which case my next question is, would our package be able to work with the older SoftDevice?

aykevl, 2020-08-29:
You might be able to get away with pretending the device is s132v6, but it's not a supported configuration. To use the nRF52 bootloader (with s140v6, or S140 version 6.1.1) support must be added for this specific configuration.

aykevl, 2020-08-29:
I'm working on support, it doesn't look very difficult.

aykevl, 2020-08-29:
See: https://github.com/tinygo-org/tinygo/pull/1329 and https://github.com/tinygo-org/bluetooth/pull/10.

deadprogram, 2020-09-02:
The needed code for this was merged into Bluetooth package here https://github.com/tinygo-org/bluetooth/pull/10 and will be part of next release. Note that a corresponding release of TinyGo itself will also be required.

deadprogram, 2020-09-17:
This has now been released, so closing.

closed 2020-09-17 by deadprogram