tinygo-org/bluetooth #15: sd: unable to run scanner example

deadprogram, 2020-09-07:
When I try to run the scanner example with `-target circuitplay-bluefruit`, the call to `adaptor.Scan()` results in a crash. The message itself seems to be obscured, nothing comes thru except the character `p` which I perhaps assume might be due to a `panic` getting lost?

aykevl, 2020-09-09:
This is what I think should work:

```diff
diff --git a/src/machine/usb_nrf52840.go b/src/machine/usb_nrf52840.go
index c8375619..e5132335 100644
--- a/src/machine/usb_nrf52840.go
+++ b/src/machine/usb_nrf52840.go
@@ -87,9 +87,12 @@ func exitCriticalSection() {
 
 // Configure the USB CDC interface. The config is here for compatibility with the UART interface.
 func (usbcdc *USBCDC) Configure(config UARTConfig) {
-       // enable IRQ
+       // Enable IRQ. Make sure this is higher than the SWI2 interrupt handler so
+       // that it is possible to print to the console from a BLE interrupt. You
+       // shouldn't generally do that but it is useful for debugging and panic
+       // logging.
        usbcdc.interrupt = interrupt.New(nrf.IRQ_USBD, USB.handleInterrupt)
-       usbcdc.interrupt.SetPriority(0xD0)
+       usbcdc.interrupt.SetPriority(0x40) // interrupt priority 2 (lower number means more important)
        usbcdc.interrupt.Enable()
 
        // enable USB
```

However, I can't seem to get USB-CDC working reliably on the nrf52840 chips. @deadprogram can you try whether it works for you? Flashing works, so it can't be an issue with the cable.

deadprogram, 2020-09-09:
The flashing works if you are doing non-Bluetooth things. Testing with the normal "examples/echo" will show that.

I think the issue with flashing is that once board crashes, so does the USB CDC code, and you have to double-tap to get to the bootloader and use `tinygo build`.

In any case, changing the `SetPriority()` value does not seem to change the behavior.

aykevl, 2020-09-09:
Ah I see, I forgot to add a `time.Sleep` at the start to see the first output (I did it to an example I tested before, but not the scanner example). I can now again reproduce the issue.

aykevl, 2020-09-09:
Well here you go, this is with my proposed interrupt priority fix:

```
scanning...
panic: runtime error: nil pointer dereference
```

Probably the same as https://github.com/tinygo-org/bluetooth/pull/14#pullrequestreview-485059240.

deadprogram, 2020-09-14:
This is working now, marking for next release.

deadprogram, 2020-09-17:
This has now been released, so closing.

closed 2020-09-17 by deadprogram