tinygo-org/bluetooth #28: Arduino compat

EwanValentine, 2020-09-19:
Is this library compatible with Arduino's at all? Apologies if this is a silly question!

aykevl, 2020-09-20:
Do you have a specific Arduino in mind?

If you mean the Arduino Uno, then no. At the moment we only support systems that have built-in Bluetooth support. I would like to add support for at least one external networking (BLE) chip. Unfortunately some commonly used boards (such as the HM-10) are so inflexible that they are hard to impossible to support in this library.

EwanValentine, 2020-09-20:
@aykevl A Uno in my case, I've got a Grove Shield and a Grove serial Bluetooth module.  

Ahh gotcha, yeah that makes sense, yeah this isn't integrated, effectively just an I/O 'thing'.

Thanks anyway, keep up the awesome work!

closed 2020-09-20 by EwanValentine

aykevl, 2020-09-20:
Yes, this shield appears to only support serial over Bluetooth Classic. That's not something this library supports right now, but might support in the future (we don't support BT classic right now).

EwanValentine, 2020-09-20:
@aykevl gotcha, thanks! Is there a suggested way of transmitting data to and from an Arduino using TinyGo at all? Could I use a Wifi shield maybe? Appreciate any pointers!

aykevl, 2020-09-20:
I guess that depends on what you want to do with it. I haven't personally done this, I prefer BLE connections with Nordic Semiconductor chips (nrf52832, etc) and I like more powerful ARM chips.

EwanValentine, 2020-09-21:
Makes sense, cheers for the info!