tinygo-org/bluetooth #30: Add ability to disconnect device

ecksun, 2020-09-21:
There seems to be no way to disconnect a connected device.

My current use-case is to develop testing software that runs on a linux machine and connects to a bluetooth server.

For that it would be very useful to have a function:

```go
func (a *Adapter) Disconnect(address Addresser) error
```

Following that it will probably be useful to also be able to list connected devices, however I will leave that for a future issue :)

deadprogram, 2020-09-21:
I would expect the API to look more like this:

```go
func (d *Device) Disconnect()
...

Since presumably you would use like this

```go
device, _ := adapter.Connect()

// do something with device

device.Disconnect()
```

aykevl, 2020-09-21:
Yes, I agree with @deadprogram here. You want to disconnect a device, so it should be a method on a device. Also, some BLE stacks may forget about the address after the connection is established (such as the SoftDevice) so that the `Device` object is necessary to keep track of the connection handle.

> Following that it will probably be useful to also be able to list connected devices, however I will leave that for a future issue :)

This may be possible on some BLE stacks, but I'm pretty sure the SoftDevice (in nrf chips) does not have an API for this. In other words, you'd need to keep track of this yourself.

On Linux (with BlueZ), if a device is already connected no attempt to reconnect is made, the existing connection handle is returned. Connected devices are also returned in the `Scan` function (even though they're usually not actively broadcasting while connected) so you can easily connect to an already connected device.

ecksun, 2020-09-21:
TL;DR: Sounds good :)

> You want to disconnect a device, so it should be a method on a device

That was my first thought but since I also wanted the ability to disconnect devices without a device handle that wouldn't have worked.

> This may be possible on some BLE stacks, but I'm pretty sure the SoftDevice (in nrf chips) does not have an API for this. In other words, you'd need to keep track of this yourself.

Aha, I see

 > On Linux (with BlueZ), if a device is already connected no attempt to reconnect is made, the existing connection handle is returned. Connected devices are also returned in the Scan function (even though they're usually not actively broadcasting while connected) so you can easily connect to an already connected device.

Ah, I see, thats nice and would cover most cases I guess.

But then if I run a program and send `SIGKILL` to it I would have to use `bluetoothctl` or similar to figure out which devices to "connect" and disconnect from

ecksun, 2020-10-15:
Oh, I must have forgotten to close this, this is obviously implemented and thank you very much for that :)

closed 2020-10-15 by ecksun