tinygo-org/bluetooth #31: adapter.DefaultAdvertisement undefined

andarm, 2020-09-22:
adapter.DefaultAdvertisement undefined (type *bluetooth.Adapter has no field or method DefaultAdvertisement)

deadprogram, 2020-09-22:
What OS and program are you trying to compile, @andarm

SpringHawk, 2020-10-09:
I have the same issue on MacOS, using Goland and trying to flash the advertisement example onto a feather-nrf52840. 
The package is installed in my GOPATH alongside the project where I want to use it.
To install it I used `go get tinygo.org/x/bluetooth`.

andarm, 2020-10-11:
> What OS and program are you trying to compile, @andarm

Window10

deadprogram, 2020-10-11:
@andarm please see https://github.com/tinygo-org/bluetooth#windows

> The Windows support is still experimental, and needs additional development to be useful. At this time, it can only be used to perform scanning operations as a BLE Central.

deadprogram, 2020-10-11:
@SpringHawk you should probably open a separate issue.

deadprogram, 2020-10-28:
Windows support is being discussed in #13 so I am closing this as duplicate. Thanks!

closed 2020-10-28 by deadprogram