tinygo-org/bluetooth #35: Field not found: Roles

Feresey, 2020-09-26:
```text
$ go run ./examples/scanner
panic: failed to enable BLE stack: MapToStruct: Field not found: Roles

goroutine 1 [running]:
main.must(...)
	/dev/shm/bluetooth/examples/scanner/main.go:23
main.main()
	/dev/shm/bluetooth/examples/scanner/main.go:11 +0x1df
```

os: Archlinux
kernel: 5.8.10
bluez: 5.55
go:  1.15.2

Feresey, 2020-09-26:
Maybe go-bluetooth does not support bluez version 5.55. The latest version in the repo is 5.54.

Feresey, 2020-09-26:
Please update go-bluetooth to latest version

https://github.com/muka/go-bluetooth/commit/4ca7d8dd0ff58cbe28676bcde4bd76939aef6808

deadprogram, 2020-09-27:
Please try out https://github.com/tinygo-org/bluetooth/tree/update-go-bluetooth-linux branch which included the updated module. Thanks @Feresey

deadprogram, 2020-11-22:
This has now been released with v0.2.0 thanks!

PR #36 (2020-10-03): modules: update to latest version of go-bluetooth package for Linux
This PR updates to the latest version of go-bluetooth package for Linux. Should fix #35
closed 2020-11-22 by deadprogram