tinygo-org/bluetooth #39: vendoring not possible because of missing dependencies

ecksun, 2020-10-08:
I cannot seem to vendor my dependency in any project with a dependency on
tinygo.org/x/bluetooth.

I can't run `go mod tidy` with the same error as show below.

I get the feeling it is related to the tooling not understanding the build tags as the errors appear to be related to importing tinygo packages.

# Steps to reproduce

```
$ go mod init example.test/foobar
go: creating new go.mod: module example.test/foobar
$ wget --quiet https://raw.githubusercontent.com/tinygo-org/bluetooth/release/examples/scanner/main.go
$ go run ./main.go 
go: finding module for package tinygo.org/x/bluetooth
go: found tinygo.org/x/bluetooth in tinygo.org/x/bluetooth v0.1.0
scanning...
found device: ...
^Csignal: interrupt
$ go mod vendor
example.test/foobar imports
        tinygo.org/x/bluetooth imports
        device/arm: package device/arm is not in GOROOT (/home/user/bin/go1.15.1.linux-amd64/src/device/arm)
example.test/foobar imports
        tinygo.org/x/bluetooth imports
        device/nrf: package device/nrf is not in GOROOT (/home/user/bin/go1.15.1.linux-amd64/src/device/nrf)
example.test/foobar imports
        tinygo.org/x/bluetooth imports
        machine: package machine is not in GOROOT (/home/user/bin/go1.15.1.linux-amd64/src/machine)
example.test/foobar imports
        tinygo.org/x/bluetooth imports
        runtime/interrupt: package runtime/interrupt is not in GOROOT (/home/user/bin/go1.15.1.linux-amd64/src/runtime/interrupt)
example.test/foobar imports
        tinygo.org/x/bluetooth imports
        runtime/volatile: package runtime/volatile is not in GOROOT (/home/user/bin/go1.15.1.linux-amd64/src/runtime/volatile)
```

deadprogram, 2020-10-24:
I'm not sure that we are going to be able to handle module vendoring at the moment with the build tags that are needed to support TinyGo and various hardware platforms. I'll play around and see if I can figure something out.

aykevl, 2020-10-24:
There might be a way to do this.

First, look up the GOROOT that is used by TinyGo. For example, using:

```
~$ tinygo info microbit
[...]
cached GOROOT:     /path/to/tinygo/goroot
```

Then you can use the GOROOT during the vendor operation:

```
GOROOT=/path/to/tinygo/goroot go mod vendor
```

ecksun, 2020-10-25:
> I'm not sure that we are going to be able to handle module vendoring at the moment with the build tags that are needed to support TinyGo and various hardware platforms. I'll play around and see if I can figure something out.

I see, I guess we can leave this issue open until it is, because it is a rather useful feature.

> First, look up the GOROOT that is used by TinyGo. For example, using:

Alright, I'm not using TinyGo though :(

Anyway there is nice it is possible at all :)

acouvreur, 2026-10-17:
Fixed: go mod vendor works with current Go modules.

closed 2026-10-17 by acouvreur