tinygo-org/bluetooth #41: AdvertisementPayload.Bytes() is nil on linux

ecksun, 2020-10-15:
It appears as if [AdvertisementPayload.Bytes()](https://pkg.go.dev/tinygo.org/x/bluetooth#AdvertisementPayload) is nil on linux.

What I specifically need is the [ManufacturerData](https://github.com/muka/go-bluetooth/blob/44d49b402aee74ccd4342e672223c19ca510bef6/bluez/profile/device/gen_Device1.go#L164) however I'm not sure how to fit that into the `AdvertisementPayload.Bytes()` interface.

deadprogram, 2020-10-15:
You mean you want to publish this data? `AdvertisementPayload` is for when you want to act as a peripheral.

ecksun, 2020-10-15:
> You mean you want to publish this data?

No, i want the ManufacturerData which from the advertisement of other peripherals

> `AdvertisementPayload` is for when you want to act as a peripheral.

Are you sure about that?

[Adapter.Scan](https://pkg.go.dev/tinygo.org/x/bluetooth#Adapter.Scan) takes a func(*Adapter, [ScanResult](https://pkg.go.dev/tinygo.org/x/bluetooth#ScanResult)) which contains an [AdvertisementPayload](https://pkg.go.dev/tinygo.org/x/bluetooth#AdvertisementPayload).

Obviously I can be mistaken but I couldn't find how to get the manufacturerdata when trying earlier today.

aykevl, 2020-10-25:
`AdvertisementPayload.Bytes()` will always be nil on Linux, because BlueZ does not expose this information.

That said, adding manufacturer data to the `AdvertisementPayload` should not be very difficult to do, as BlueZ does expose this information. It's just not implemented yet.

ecksun, 2020-10-25:
> AdvertisementPayload.Bytes() will always be nil on Linux, because BlueZ does not expose this information.

Alright, good to know.

> That said, adding manufacturer data to the AdvertisementPayload should not be very difficult to do, as BlueZ does expose this information. It's just not implemented yet.

Are you suggesting doing something like this:

```go
type AdvertisementPayload interface {
	LocalName() string
	HasServiceUUID(UUID) bool
	Bytes() []byte
	ManufacturerData() map[uint16]interface{} // map[uint16][]byte ?
}
```

aykevl, 2020-10-25:
Yes something like that (with map[uint16][]byte` instead of `map[uint16]interface{}`), or maybe even this:

```go
type AdvertisementPayload interface {
	LocalName() string
	HasServiceUUID(UUID) bool
	ManufacturerData(key uint16) []byte
	Bytes() []byte
}
```

acouvreur, 2026-10-17:
Fixed by #100.

closed 2026-10-17 by acouvreur