tinygo-org/bluetooth #48: macos Big Sur (11.0.0) - INFO

lacendarko, 2020-12-21:
Hi, I would like to point out a problem I had with upgrading to macOS 11 and constantly getting an error "abort trap". Go to System Preferences -> Security & Privacy -> Privacy and add iTerm (or I guess Terminal if you're not using iTerm).

antonfisher, 2021-01-19:
@lacendarko i think you meant "System Preferences" —> "Security & Privacy" —> "**Bluetooth**" 👍

Gys, 2021-01-25:
For me it was a combination of both: "System Preferences" —> "Security & Privacy" —> "Privacy" -> "Bluetooth"

gedw99, 2021-09-04:
Not working on MAC for me btw...

If there is anything i can do to help debug let me know...

I enabled bluetooth access to the scanner binary example.

[image]

``` 
# https://github.com/tinygo-org/bluetooth

print:

hack:
	# manually give permisssions on mac
	# https://github.com/tinygo-org/bluetooth/issues/48

dep:
	rm -rf bluetooth
	git clone https://github.com/tinygo-org/bluetooth -b dev

EX_FOLDERPATH=bluetooth/examples/scanner
EX_BIN_FOLDERPATH=$(PWD)/_dist
EX_BIN_NAME=scanner
EX_BIN=$(EX_BIN_FOLDERPATH)/$(EX_BIN_NAME)
ex-buildrun:
	cd $(EX_FOLDERPATH) && go build -o $(EX_BIN) .
	cd $(EX_BIN_FOLDERPATH) && ./$(EX_BIN_NAME)
```  

``` 
make ex-buildrun
cd bluetooth/examples/scanner && go build -o /Users/apple/workspace/go/src/github.com/gedw99/gio-proposals/bluetooth/providence/tinygo-org__bluetooth/_dist/scanner .
cd /Users/apple/workspace/go/src/github.com/gedw99/gio-proposals/bluetooth/providence/tinygo-org__bluetooth/_dist && ./scanner
/bin/sh: line 1: 51213 Abort trap: 6           ./scanner
make: *** [ex-buildrun] Error 134
```

env:

``` 
go version
go version go1.17 darwin/amd64

go env
GO111MODULE="on"
GOARCH="amd64"
GOBIN=""
GOCACHE="/Users/apple/Library/Caches/go-build"
GOENV="/Users/apple/Library/Application Support/go/env"
GOEXE=""
GOEXPERIMENT=""
GOFLAGS=""
GOHOSTARCH="amd64"
GOHOSTOS="darwin"
GOINSECURE=""
GOMODCACHE="/Users/apple/workspace/go/pkg/mod"
GONOPROXY=""
GONOSUMDB=""
GOOS="darwin"
GOPATH="/Users/apple/workspace/go"
GOPRIVATE=""
GOPROXY="https://proxy.golang.org,direct"
GOROOT="/usr/local/opt/go/libexec"
GOSUMDB="sum.golang.org"
GOTMPDIR=""
GOTOOLDIR="/usr/local/opt/go/libexec/pkg/tool/darwin_amd64"
GOVCS=""
GOVERSION="go1.17"
GCCGO="gccgo"
AR="ar"
CC="clang"
CXX="clang++"
CGO_ENABLED="1"
GOMOD="/dev/null"
CGO_CFLAGS="-g -O2"
CGO_CPPFLAGS=""
CGO_CXXFLAGS="-g -O2"
CGO_FFLAGS="-g -O2"
CGO_LDFLAGS="-g -O2"
PKG_CONFIG="pkg-config"
GOGCCFLAGS="-fPIC -arch x86_64 -m64 -pthread -fno-caret-diagnostics -Qunused-arguments -fmessage-length=0 -fdebug-prefix-map=/var/folders/wp/ff6sz9qs6g71jnm12nj2kbyw0000gp/T/go-build1305902903=/tmp/go-build -gno-record-gcc-switches -fno-common"

```

deadprogram, 2021-09-04:
TinyGo 0.19 is not compatible with Go 1.17.  Please see https://github.com/tinygo-org/tinygo/issues/2080

idefixcert, 2021-11-22:
I still have the problem that scanner does not work.
I also use go 1.17 but it seems the issue mentioned ago is fixed?

mo6zes, 2022-03-21:
I do experience the same issue, it does not find any devices. Have you found a solution?

aykevl, 2023-04-25:
I've added a note to the README in https://github.com/tinygo-org/bluetooth/pull/157.