tinygo-org/bluetooth #63: nrf52840: Scan not working as expected

deadprogram, 2021-05-01:
Took me a while to actually report this, because I thought it had to be some problem on my machine, but after testing on another system, I think it is something.

I am currently unable to `Scan()` from the three different nrf52840 devices that I have tried (Itsybitsy-nRF52840, Circuit Playground Bluefruit, and Clue) using either the latest `release` or `dev` branches of both TinyGo and Bluetooth.

I am able to run advertiser successfully, but not scan.

One interesting note, is that when I tested the Clue using the test program I have with the attached display, sometimes after a very long period of time (10 or more minutes) it would suddenly start to work. However this was very inconsistent.

aykevl, 2021-05-02:
I have tested the PCA10040 and Circuit Playground Bluefruit and they both work on my side.

This is the binary that I'm using on the Circuit Playground Bluefruit, that I have tested and confirmed to work:

```
~/src/tinygo.org/x/bluetooth$ tinygo build -o scanner.uf2 -target=circuitplay-bluefruit -size=short ./examples/scanner/
   code    data     bss |   flash     ram
  10252     116    4208 |   10368    4324

~/src/tinygo.org/x/bluetooth$ md5sum scanner.uf2
324c2f847d71fcd64a9afec68e036aa4  scanner.uf2
```

Here is the file: [scanner.zip](https://github.com/tinygo-org/bluetooth/files/6411621/scanner.zip) (encoded in a zip file otherwise GitHub won't allow the upload).

deadprogram, 2021-05-02:
I downloaded that file:

```
$ md5sum ~/Desktop/flash.uf2 
324c2f847d71fcd64a9afec68e036aa4  /home/ron/Desktop/flash.uf2
```

Copied to my Circuit Playground Bluefruit. Appeared to flash as expected, but no data coming from USB CDC.

BTW this is the bootloader version I have on the board:

```
$ cat /media/ron/CPLAYBTBOOT/INFO_UF2.TXT 
UF2 Bootloader 0.4.0 lib/nrfx (v2.0.0) lib/tinyusb (0.6.0-272-g4e6aa0d8) lib/uf2 (remotes/origin/configupdate-9-gadbb8c7)
Model: Adafruit Circuit Playground nRF52840
Board-ID: nRF52840-CircuitPlayground-revD
SoftDevice: S140 version 6.1.1
Date: Jan 19 2021
```

aykevl, 2021-05-02:
It appears that you have a newer bootloader, from Jan 19 2021. Mine is much older:

```
$ cat /media/ayke/CPLAYBTBOOT/INFO_UF2.TXT 
UF2 Bootloader 0.2.11-8-g2c13fd5-dirty lib/nrfx (v1.1.0-1-g096e770) lib/tinyusb (legacy-755-g55874813) s140 6.1.1
Model: Adafruit Circuit Playground nRF52840
Board-ID: nRF52840-CircuitPlayground-revD
Date: Jul 13 2019
```

Is this perhaps an older board that has an updated bootloader, or one you bought recently?

deadprogram, 2021-05-02:
Yes, I have updated the bootloader. I previous tried with several different of the most recent versions (0.4, 0.5) actually just to be sure. Did not help.

deadprogram, 2021-05-06:
This is where I obtained the updated bootloaders from: https://github.com/adafruit/Adafruit_nRF52_Bootloader/releases

deadprogram, 2021-05-18:
Additional update: this package worked within last couple of months with v0.4.0 bootloader on my ItsyBitsy nRF 52840 board. So it is not just a question of bootloader version.

aykevl, 2021-05-19:
> Additional update: this package worked within last couple of months with v0.4.0 bootloader on my ItsyBitsy nRF 52840 board. So it is not just a question of bootloader version.

Well that's interesting. In that case it might be a compiler change too.

aykevl, 2021-05-19:
How did you update the bootloader? According to the release notes, it can only be done easily when you're already on version 0.4.0 or higher:

> * Add self-update feature with update-{board}.uf2. This allow bootloader to update itself easily (requires running bootloader with at least 0.4.0)

And when I tried updating that way, nothing seemed to happen: the file just sits there on the virtual FAT drive.

deadprogram, 2021-05-19:
https://tinygo.org/docs/reference/microcontrollers/circuitplay-bluefruit/#updating-the-uf2-bootloader :smile_cat:

aykevl, 2021-05-19:
Thank you! I will try that later.

deadprogram, 2021-05-27:
Any updates on this by any chance? Or ideas on how it could be debugged?

oflebbe, 2021-07-03:
Actually Advertising in #1949 is not working correctly as well. Sometimes the intervals are 40sec rather in the 0.1sec range.

aykevl, 2021-07-05:
> Actually Advertising in [tinygo-org/tinygo#1949](https://github.com/tinygo-org/tinygo/issues/1949) is not working correctly as well. Sometimes the intervals are 40sec rather in the 0.1sec range.

I'm not sure what you mean? The issue in tinygo-org/tinygo#1949 is very different and is about a stack overflow. That _might_ also be the case here but I doubt it. In the case of this bug, it has most likely something to do with the bootloader.

oflebbe, 2021-07-05:
I found the issue: I had a device (mobile) around which tried to connect to the Heart Rate Services.  Sorry. Please disregard my comment.

aykevl, 2021-07-06:
I still can't reproduce the issue. I have updated the bootloader on my Circuit Playground Bluefruit:

```
$ cat /media/ayke/CPLAYBTBOOT/INFO_UF2.TXT
UF2 Bootloader 0.4.1 lib/nrfx (v2.0.0) lib/tinyusb (0.6.0-272-g4e6aa0d8) lib/uf2 (remotes/origin/configupdate-9-gadbb8c7)
Model: Adafruit Circuit Playground nRF52840
Board-ID: nRF52840-CircuitPlayground-revD
SoftDevice: S140 version 6.1.1
Date: Feb 17 2021
```

Then flashed the scanner example:

```
tinygo flash -target=circuitplay-bluefruit ./examples/scanner/
```

And it works as expected.

@deadprogram can you try with the exact same bootloader as I have above (0.4.1, following the link you gave me) and this [scanner.zip](https://github.com/tinygo-org/bluetooth/files/6770947/scanner.zip) binary? (Extract before use, of course - GitHub doesn't allow uploading .uf2 files). If it still doesn't work for you on the Circuit Playground Bluefruit then I'm going to suspect a hardware issue.

deadprogram, 2021-07-06:
OK this is a bit weird. The file you sent worked as expected even with bootloader 0.5.0.

Then I tested with the latest TinyGo release, and it also worked as expected.

I do not think it was any code change, must have been some problem with cached files? I am going to experiment a bit more now before closing.

deadprogram, 2021-07-06:
Able to build/run examples on other Adafruit boards now as well. Must have been some cached files.

I will close now, and reopen if needed. Thanks for your patience @aykevl

deadprogram, 2021-12-13:
Reopening because it is happening again :( 

Sometimes after a really long time (1 hour or more?) it will catch a couple of scans, but then nothing again.

aykevl, 2021-12-17:
Still works for me.

- TinyGo version: e4de7b49575c28fcd3a446e7478687e9611d110a
- Command: `tinygo build -o test.uf2 -target=circuitplay-bluefruit -size=short -print-stacks ./examples/scanner/`
- md5sum: 5c8ea52975d3991e87bf7419beb6a998 (test.uf2)

closed 2024-02-21 by deadprogram