tinygo-org/bluetooth #67: Connection not working on

andrewrynhard, 2021-06-12:
I have flashed the heart rate example to a [nRF52840-DK)](https://www.nordicsemi.com/Software-and-tools/Development-Kits/nRF52840-DK) board, and am unable to get anything to connect. It seems that they get the initial advertised services, but the connection eventually times out. Not sure what you need from me, but happy to share whatever.

aykevl, 2021-06-13:
Sadly my nrf52840-DK has died so I can't reproduce it locally.

One thing you can check: are you using the correct SoftDevice? Try `nrfjprog -f nrf52 --eraseall`, then flashing the SoftDevice (S140) and then the heart rate example. Also, what app are you using to connect? I usually nRF Connect from Nordic, which works great for debugging all kinds of BLE issues.

If you are able to connect (you can connect and see the services in nRF Connect) then you're already past like 99% of possible issues. I don't know why it would fail then. Have you looked at the serial console, to see whether it prints anything interesting?

andrewrynhard, 2021-06-13:
Here are the steps I took:

- Download and unpack the SoftDevice firmware from https://www.nordicsemi.com/Software-and-tools/Software/S140
- Flashed the SoftDevice firmware: `nrfjprog --family NRF52 --chiperase --verify --program s140_nrf52_7.0.1_softdevice.hex`
- Flashed the example: `tinygo flash -target=pca10056-s140v7`

I can see output on the serial console. I have also tried building with `tinygo build -o payload.hex -size=short -target=pca10056-s140v7 .` and then flashing with `nrfjprog --family nrf52 --sectorerase --program payload.hex --verify`. I am using nRF Connect. The discovery happens just fine, but connecting hangs:

[image]

aykevl, 2021-06-13:
Hmm, strange. So the console output doesn't show anything unexpected, like an error?

andrewrynhard, 2021-06-13:
> Hmm, strange. So the console output doesn't show anything unexpected, like an error?

It does not. Relatively new to this space, so anywhere you want me to look, just point me in the direction.

eldondev, 2021-07-24:
Could this be related to #53 ?

mlsorensen, 2023-01-15:
I know this is old, but I experience exactly the same thing. I have two of the circuitplay-bluefruit devices. When I load any examples and try to use any of the scanner apps to connect, they just hang at "connecting" and then time out.

What is interesting is that the connect handler fires. For example the lights turn on after I attempt to connect with the neopixel example. But scanners never complete a connect or see any services.