tinygo-org/bluetooth #68: 16bits UUID byte format on macos is different from other platforms.

nobonobo, 2021-06-18:
on macos:
`180a0000-0000-0000-0000-000000000000`

on others:
`0000180a-0000-1000-8000-00805f9b34fb`

specification:
128_bit_value = 16_bit_value * 2^96 + Bluetooth_Base_UUID

deadprogram, 2021-06-20:
Seems related to https://github.com/JuulLabs-OSS/cbgo/pull/3 what do you think @nobonobo?

nobonobo, 2021-07-02:
Perhaps this line will convert the UUID of unknown length to a string and parse it.
As a result, I think I got a UUID byte string that is different from other platforms.

https://github.com/tinygo-org/bluetooth/blob/28f9f4e69ea8116eb3f180ec9c050cf9d27aa82c/gattc_darwin.go#L28

May be, We will need support 32bits UUID format for JuulLabs-OSS/cbgo#3.

deadprogram, 2022-08-30:
This has now been fixed in #113 

Thanks for helping out @nobonobo 

I will marked it to be closed after the next release.

deadprogram, 2022-10-27:
The fix for this has been released, so now closing. Thank you!

PR #113 (2022-09-05): macos: correct handling for 16 bit UUIDs
This PR fixes #68 by switching to the tinygo-org fork of the cbgo package that includes both the @gkuchta fix as well as some others related to correctly parsing UUIDs on macOS.

As a result, the `New16BitUUID()` function can now be unified across all platforms.
closed 2022-10-27 by deadprogram