tinygo-org/bluetooth #70: Support Arduino nina-fw

bgould, 2021-07-18:
It appears that if you pull the right pin low when an ESP-32 module with the Arduino nina-fw (e.g., Arduino Nano IOT, Adafruit Metro M4 Airlift, Pyportal, etc) it will put the module into Bluetooth mode at startup:

https://github.com/arduino/nina-fw/blob/fc6bd42754ecf8e32079f9f2f3bec22d177ce0bf/main/sketch.ino.cpp#L108

it appears the same functionality also exists in the Adafruit fork:
https://github.com/adafruit/nina-fw/blob/0e267bc8858f188f5ac7c1527d0554d8294ca756/main/sketch.ino.cpp#L95

If the Bluetooth mode is enabled, instead of processing wifinina commands over SPI, the ESP32 instead will process BLE related commands over UART.  I have not found documentation on the protocol, but it might be possible to distill it from the ArduinoBLE library :

https://github.com/arduino-libraries/ArduinoBLE/tree/master/src/utility

bgould, 2023-12-21:
see https://github.com/tinygo-org/bluetooth/pull/207

deadprogram, 2024-03-24:
Now closing since 0.9.0 has been released. Thank you!

closed 2024-03-24 by deadprogram