tinygo-org/bluetooth #71: Issue advertising device name > 8 characters on adafruit nrf52840 feather

eldondev, 2021-07-24:
I am attempting to run a modification of the nusserver example on an adafruit nrf52840 feather. It works when the size of the LocalName advertised is 8 characters or less. When more than 8 characters, it does not seem to work, as scanning never locates the name. I will try to collect additional logging. When running identical code on a regular linux host, it seems to work.

aykevl, 2021-07-25:
Did you check whether the `Configure` method returns an error?

eldondev, 2021-07-26:
Yes, it did! I was overflowing the bounds of the advertisement packet size, with the addition of the service uuid I was advertising. Since I control both client and server, I just advertise the device name, since I already know about the services. Thanks!

closed 2021-07-26 by eldondev