tinygo-org/bluetooth #76: AdapterExists: 'Rejected send message' via Chrome OS Debian Linux

khaelou, 2021-08-13:
I followed the guide considering installation on Linux however I'm getting this error when running example:

[image]

Any suggestions? I'm sure this isn't a bug yet I'm having a hard time understanding why such panic occurred.

moderateroni, 2021-12-07:
@khaelou , try running your go program with sudo.. perhaps you need to use the absolute path for your go binary, but then it should work... something like sudo /usr/local/bin/go run main.go

khaelou, 2021-12-19:
@moderateroni Will try this, not really an issue so I'll close this 👍

closed 2021-12-19 by khaelou