tinygo-org/bluetooth #78: is it useable for android?

MatejMagat305, 2021-09-11:
Hi I m coding in https://github.com/fyne-io/fyne, and it can run on android, it this library suitable for android?

deadprogram, 2021-09-18:
Hello @MatejMagat305 not at this time. I will look into what would be required, but seems like it would take quite a bit of work.

gedw99, 2021-09-20:
I was going to ask exactly the same thing. 

I am building hardware with haptics and ble, and from the APP need to communicate it. I use GIO not fyne.

MatejMagat305, 2021-09-20:
> I was going to ask exactly the same thing.
> 
> I am building hardware with haptics and ble, and from the APP need to communicate it. I use GIO not fyne.

it never mind wheter fyne or gio(although in my opinion it fyne more clear, but it is only my.......), mainly it will wonderfull when this library will suport android maybe the authors might also think about iOS, but I think the android is more comon and opensource........

MatejMagat305, 2021-12-28:
I  was trying (my experimental library: [bluetooth](https://github.com/MatejMagat305/bluetooth_with_fyne_mobile/tree/main/fyne_bluetooth/bluetoothSocket) and after that I was searching how it is on android and it will be able after implemention of permision therefore I create isues: https://github.com/golang/go/issues/50104. Although [jni in android](https://android.googlesource.com/platform/frameworks/base/+/jb-release/core/jni/android_bluetooth_BluetoothSocket.cpp) show own implementation of socket, it still use `#include <sys/socket.h>`, which I suppose is (in golang implementation) in unix package,  what do you think?

stevekuznetsov, 2022-03-26:
@deadprogram is there a good doc to understand what's different about the Android BLE stack to get a better idea of what would be necessary to support that in this library? I would like to write a cross-platform app using BLE but am stuck without support for BLE on Android in Go. If It's something accessible, I might be able to work on this support for the library here.

MatejMagat305, 2022-03-26:
@stevekuznetsov I was trying implement it like socket (above), but now I think we need resolve: https://github.com/golang/go/issues/50104, and maybe only way how implement it is jni, I want do it in summer, do you join to me?

stevekuznetsov, 2022-03-26:
@MatejMagat305 I am not very familiar with the system so I'd have to understand more about how Android wants to handle BLE first, but maybe!

MatejMagat305, 2022-03-27:
@stevekuznetsov so, I m not big expert to Android OS, but .............., I would like have bluetooth access from go ............

MatejMagat305, 2023-04-23:
maybe it is already implemented in https://git.wow.st/gmp/ble

Sakura0721, 2023-06-04:
Have you made it work on android?

MatejMagat305, 2023-06-04:
> Have you made it work on android?

well yes, but only for fyne and it is nit tested (it could be easy to port to other library which provide env, vm and ctx like gomobile) https://github.com/fyne-io/fyne-x/pull/58

Solander, 2023-09-02:
> > Have you made it work on android?
> 
> well yes, but only for fyne and it is not (until now) tested, it could be easy to port to other library which provide env, vm and ctx like gomobile: [fyne-io/fyne-x#58](https://github.com/fyne-io/fyne-x/pull/58)

I guess it takes some work, but it would be amazing if this was a working library that was released! I'm trying to do exactly this with Fyne, so I will take a closer look at your PR and see if I can try it.
Great job! 
Hopefully it will help tinygo as well. I like that interface for interacting with Bluetooth.