tinygo-org/bluetooth #102: Cannot compile in windows

tmtong, 2022-06-11:
I install cygwin in windows 10

```
export GOARCH=amd64
export GOOS=windows
git clone https://github.com/tinygo-org/bluetooth.git

bash:/tmp/bluetooth$ go build ./examples/scanner/main.go
# tinygo.org/x/bluetooth/winbt
winbt\advertisement.go:59:11: undefined: NewEvent
winbt\advertisement.go:59:87: undefined: Event
winbt\advertisement.go:75:11: undefined: NewEvent
winbt\advertisement.go:75:87: undefined: Event
```

It seems to work fine in linux

bartmeuris, 2022-07-10:
I got it working with the 64bit version of msys2 ( https://www.msys2.org/ ) - but windows support is currently very lacking, it only supports some (very basic) scanning at the moment.

deadprogram, 2022-07-11:
Please see https://github.com/tinygo-org/bluetooth/issues/13

deadprogram, 2024-02-21:
This should now be corrected, so now closing.

closed 2024-02-21 by deadprogram