tinygo-org/bluetooth #118: Fail on connect: Properties.GetAll org.bluez.Device1: Method "GetAll" with signature "s" on interface "org.freedesktop.DBus.Properties" doesn't exist

aj3423, 2022-09-16:
It's strange that yesterday it still works, but today it shows error when connecting to a bluetooth device:

> Properties.GetAll org.bluez.Device1: Method "GetAll" with signature "s" on interface "org.freedesktop.DBus.Properties" doesn't exist

The code:
```golang
var adapter = bluetooth.DefaultAdapter
if e := adapter.Enable(); e != nil {
	panic(e)
}

mac, _ := bluetooth.ParseMAC("71:42:F7:76:BC:88") 
a := bluetooth.Address{MACAddress: bluetooth.MACAddress{MAC: mac}}
_, e := adapter.Connect(a, bluetooth.ConnectionParams{})
if e != nil {
	fmt.Println(e.Error())
}
```
I tried to reboot my machine, still no luck. But it can be connected from `bluetoothctl` -> `connect <MAC>`.

I'm on Kali Linux, How to trace the problem? Thanks.

deadprogram, 2022-09-16:
Did you happen to update your Bluez by any chance? Sounds a lot like it.

aj3423, 2022-09-16:
@deadprogram I checked `/var/log/apt/history.log`, I didn't upgrade in 48 hours, but I did hibernate the PC yesterday and restored it this morning. I thought maybe the hibernation causes it so I tried reboot, but problem still exists.

I upgrade the system just now, there is an upgrade for *bluetooth*, *bluez*, now it shows another error: 
> MapToStruct: Field not found: Bonded

andrewhowdencom, 2023-01-26:
Hola lovely people, 

I was personally experiencing this also. The issue somehow appears to be down to the address component (`a` in your case). Setting the bytes manually:

```
bluetooth.Address{
  bluetooth.MACAddress{
    bluetooth.MAC{
       0x71,
       0x42
       ...
    }
  }
}
```

Worked for me. I'm not sure why just yet. Anyway, without doing this `dbus` is looking up a device like:

`/dev/hci0/00_00_00_00_00_00`

(or whatever) and it does not work!

ThomasHabets, 2023-10-07:
> Did you happen to update your Bluez by any chance? Sounds a lot like it.

Is that a problem? Does this repo not support newer Bluez?

I was already doing the supposed "seems fine" way of parsing addresses, above.

ThomasHabets, 2023-10-07:
Ok, thanks to some forum posts I'm inferring that this error message in this context means something like "no recent scan was performed".

If I do a quick bluetooth scan and immediately try to connect, then it works.

If I then wait a few seconds, connects will stop working again.

deadprogram, 2024-02-21:
This should now be fixed on the `dev` branch. I will tag this issue to be closed on next release.

deadprogram, 2024-03-24:
Now closing since 0.9.0 has been released. Thank you!

closed 2024-03-24 by deadprogram