tinygo-org/bluetooth #119: Windows Implementation needs to use ReadValueWithCacheModeAsync

KeganHollern, 2022-09-23:
Hello,

I am testing on the dev branch with an [Aranet4](https://aranet.com/products/aranet4/) device. 

The my sample code for reading characteristic values works from Linux and OSX, but the windows implementation is not functioning. 

I believe this [stackoverflow](https://stackoverflow.com/questions/63336568/bluetooth-le-characteristic-values-only-read-once) post explains the problem. By using the `ReadWithAsync` function exposed by winrt-go, you're using the windows cache. My BT device does not send notifications on value change, so this cache is never updated.

I need a way to specify that the read should **not** use the cache & should instead read directly from the device.

Please fix before the windows support is added to Release! 🙏

deadprogram, 2022-09-24:
@jagobagascon have you seen this?

jagobagascon, 2022-09-26:
This is something we encountered on service and characteristic discovery. And fixed it disabling the cache on those methods:
https://github.com/tinygo-org/bluetooth/blob/8fae59774538b7110f95849c79cc9bde6c164459/gattc_windows.go#L35
https://github.com/tinygo-org/bluetooth/blob/8fae59774538b7110f95849c79cc9bde6c164459/gattc_windows.go#L152

And somehow we missed the read method. I added a PR to fix this #120

deadprogram, 2022-10-27:
This was released in v0.6.0 so now closing. Thanks, everyone!

PR #120 (2022-09-26): windows: disable cache when reading characteristics
The Windows cache may cause problems with characteristics that do not support notifications and require an explicit read. 
Checkout #119. 

This PR disables cache when reading characteristic values. 

fixes #119
closed 2022-10-27 by deadprogram