tinygo-org/bluetooth #126: Implement reconnects when device disconnects

Arsen6331, 2022-10-21:
Currently, there doesn't seem to be a way to reconnect to a device if it disconnects when acting as a central. This could either be implemented in the library itself, or OnConnect and OnDisconnect callbacks can be added, allowing callers to implement it themselves.

deadprogram, 2022-10-26:
Having a disconnect handler seems like a most excellent idea!

TirelessDev, 2023-02-24:
I have an implementation of this in #149. Tested on Linux and MacOS. I don’t have ready access to a Windows machine so haven't been able to to that yet. 

I think having the application handle reconnecting it the correct way forwards as there are many reasons a disconnect could occur that will potentially be application dependent. Having the library take care of it also adds a lot of complexity.

aykevl, 2023-05-02:
I agree with @TirelessDev here. It's best to let the application handle this.
I will take a look at #149 later.

chamabreu, 2023-05-19:
Hi, sorry to open this again, but i am struggling with this.

I have implemented this package (thanks, works great so far), but the SetConnectHandler only fires on connect events, not on disconnects.

I am on Linux.
Does not matter if i disconnect through code with the .Disconnect() function, or turning the Device off, or disconnecting over linux itself.

If the connection is established, the handler triggers.

(I am on the newest version 0.6)

TirelessDev, 2023-05-20:
Hey @chamabreu, have you tried the dev branch? The changes in #149 were merged recently and should help with this.

chamabreu, 2023-05-21:
Meh, sorry my fault.
I did not realized that the PR was not in a release.

I switched the package because I needed better reconnection handling, so I don't need it anymore to test.
But thanks for the fast response and the information I needed!

deadprogram, 2024-01-07:
Now closing since this feature was already released. Thanks everyone!

closed 2024-01-07 by deadprogram