tinygo-org/bluetooth #128: adapter.StopScan() not thread-safe

okvorb, 2022-11-11:
According to examples StopScan() must be called from Scan() callback but it is not always  convinient way to stop a scanner. It will be nice to have an option to call StopScan() method from another goroutine thread safely. Current implementation leads to data race warning on MacOS (but not on Ubuntu/Linux) on next line inside StopScan() method :

`if a.scanChan == nil {`

_a.scanChan_ variable is not synchronized for reading and writing from different goroutines what leads to the data race.

alexander-eroma, 2024-02-02:
Do you plan any activity regarding this issue?

aykevl, 2024-02-17:
@okvorb @alexander-eroma would any of you like to make a PR to fix this issue? (I'm not using macOS that often).