tinygo-org/bluetooth #129: DefaultAdapter not declared by package bluetooth compiler (UndeclaredImportedName)

mytechnotalent, 2022-11-20:
I wired up a HC-05 to a Raspberry Pi Pico's UART0.  I connect the 3.3 to 3.3 and gnd to gnd and tx to GP1 and rx to GP0.

When working with the stock example:
```
package main

import (
	"time"

	"tinygo.org/x/bluetooth"
)

var adapter = bluetooth.DefaultAdapter

func main() {
	// Enable BLE interface.
	must("enable BLE stack", adapter.Enable())

	// Define the peripheral device info.
	adv := adapter.DefaultAdvertisement()
	must("config adv", adv.Configure(bluetooth.AdvertisementOptions{
		LocalName: "Go Bluetooth",
	}))

	// Start advertising
	must("start adv", adv.Start())

	println("advertising...")
	for {
		// Sleep forever.
		time.Sleep(time.Hour)
	}
}

func must(action string, err error) {
	if err != nil {
		panic("failed to " + action + ": " + err.Error())
	}
}
```

The error:
```
DefaultAdapter not declared by package bluetooth compiler (UndeclaredImportedName)
```

Is there additional setup necessary?  There is nothing in the example showing how to handle the `DefaultAdapter` as the `bluetooth` mod on line 9 does not recognize it.

I am simply looking to run this example to connect to a central.

jfelixfalcon, 2022-11-20:
I'm dealing with the same error. For now, you can clone the repo and add it to your local packages. That is what I did to solve the problem. I recommend creating an alias:
import (
    localBluetooth "backend/pkg/bluetooth"
    ...
)

mytechnotalent, 2022-11-20:
Thanks @jfelixfalcon.  The issue is I am not making a central but a peripheral for the MCU, specifically the PICO.  I can't even get it to compile as the code does not recognize the `DefaultAdapter` and it is unclear in the docs how to get even a basic setup working.

mytechnotalent, 2022-11-27:
@aykevl can you provide an example on how we can use this?  

```var adapter = bluetooth.DefaultAdapter```

I have both a Win10 TinyGo and MAC intel and MAC M1 setup for TinyGo.  None of the three are working with the example.  I am using a HC-11.  

I would love to use TinyGo with bluetooth but just need to get an understand here as the `DefaultAdapter` is not obvious to populate properly.

deadprogram, 2022-11-27:
In order to use a device like the HC-11 on Linux, you would need to have a Linux kernel driver that let you use it from BlueZ. A brief search of the internets did not turn up anything immediately on that.

As far as using it bare metal, the support for the AT command set on that device would need to be added for it to work. At present, we only have support for the Nordic Semi stack.

Hope that helps!

mytechnotalent, 2022-11-27:
The issue @deadprogram is getting it to compile. 
```DefaultAdapter not declared by package bluetooth compiler (UndeclaredImportedName)```

aykevl, 2022-12-12:
@mytechnotalent The Bluetooth package doesn't support external devices yet. You can see the list of supported devices [in the README](https://github.com/tinygo-org/bluetooth#current-support): Linux, MacOS, Windows, and Nordic Semi chips. That is, only systems with built-in support for Bluetooth (either via the OS or directly built into the hardware). External modules aren't supported yet.

In addition:

 - HC-05 is Bluetooth Classic only, not BLE. This package only supports BLE at this time (the main commonality between BLE and Bluetooth Classic is the name, the system itself is totally different). We might support Bluetooth Classic some time in the future though.
 - The HC-11 is not a Bluetooth module so will never be supported by this package.

mytechnotalent, 2022-12-12:
thank you @aykevl I will close this out.  Appreciate the info.

closed 2022-12-12 by mytechnotalent