tinygo-org/bluetooth #130: Windows API issue with Nordic BLE

miklaine, 2022-12-02:
While integrating a 3rd party BLE device on Windows 10, I encountered a repeated crash after 3-4 minutes of data transmission. 
The device continues to work normally, but the app crashes.
Based on the logs, it appears that the issue is coming from the CGO side. The crash also happens when the app's memory footprint reaches ~32.5MB.

Any help is appreciated.

Here's the source code

```
package main

import (
	"encoding/hex"
	"fmt"
	"strings"
	"time"
	"tinygo.org/x/bluetooth"
)

var bleAdapter = bluetooth.DefaultAdapter

var startCmd = []byte{0x68, 0x68, 0x97, 0x03, 0x03, 0x01, 0x00, 0x01}
var stopCmd = []byte{0x68, 0x68, 0x97, 0x03, 0x03, 0x01, 0x00, 0x00}

func main() {

	bleAdapter.Enable()

	go bleSearch()

	select {}
}

var dev *bluetooth.Device
var srvc bluetooth.DeviceService
var rxChar bluetooth.DeviceCharacteristic
var txChar bluetooth.DeviceCharacteristic

func bleSearch() {
	ch := make(chan bluetooth.ScanResult, 1)

        err := bleAdapter.Scan(func(adapter *bluetooth.Adapter, result bluetooth.ScanResult) {
	        if strings.Contains(result.LocalName(), "RF-BLE-") {
		        fmt.Println("found device:", result.Address.String(), result.RSSI, result.LocalName())
		        ch <- result
		        bleAdapter.StopScan()
	        }
        })
        if err != nil {
	        fmt.Println(err)
        }
	
	select {
	case result := <-ch:

		dev, err = bleAdapter.Connect(result.Address, bluetooth.ConnectionParams{})
		if err != nil {
			fmt.Println(err)
			return
		}

		var srvcs []bluetooth.DeviceService
		var chars []bluetooth.DeviceCharacteristic

		srvcs, err = dev.DiscoverServices([]bluetooth.UUID{bluetooth.ServiceUUIDNordicUART})
		if err != nil {
			fmt.Println(err)
			return
		}
		if len(srvcs) == 0 {
[... 279 lines cut ...]
runtime.goexit()
        C:/Users/miklaine/go/go1.19/src/runtime/asm_amd64.s:1594 +0x1 fp=0xc0004a1fe8 sp=0xc0004a1fe0 pc=0x462fc1
created by runtime.gcBgMarkStartWorkers
        C:/Users/miklaine/go/go1.19/src/runtime/mgc.go:1159 +0x25
rax     0x1a481e0
rbx     0x191740
rcx     0x1aa6a0
rdi     0x6b5040
rsi     0xc000090b70
rbp     0xc00005fe20
rsp     0xc00005fde0
r8      0x4fe43d
r9      0x70b338
r10     0xc0000421a0
r11     0x4047ef
r12     0xc00005fdc0
r13     0xc00004c000
r14     0xc000042340
r15     0x0
rip     0xc000090b70
rflags  0x10206
cs      0x33
fs      0x53
gs      0x2b
```

gotoel, 2023-01-22:
Experiencing the same on Windows 10 22H2 (19045.2486) with similar usage. It seems to be happening after a certain amount of data is transmitted. Also sometimes I don't get a crash, it just stops receiving data.

Rabbit0w0, 2023-02-13:
Similar problem here. My program kept crashing after starting sending data

spreatty, 2023-12-22:
Same is for me. My app connects to 2 Switchbots, subscribes to notifications and writes to each device at least once per 2 minutes (to keep connection alive). Crashes after 2-7 minutes without panicing, like ExitProcess is called inside WinRT, it only prints "Process xxx has exited with status 0xc0000005".

But sometimes the app doesn't crash, it stops receiving notifications instead. And if I don't enable notifications at all it just works fine (a poor way out though).

aykevl, 2024-03-12:
Have you checked whether the latest release of the bluetooth package fixes this issue? It may have been fixed in https://github.com/tinygo-org/bluetooth/pull/208.

aykevl, 2024-03-12:
> if i set GOGC="off" (turning of garbage collector) then it stops crashing

That's very useful information! Sounds like it might be a finalizer that incorrectly frees some memory.
@jagobagascon any idea what might be wrong here?

jagobagascon, 2024-03-13:
@aykevl I think we are mixing different problems here. Setting `GOGC="off"` does not help with the error we are seeing in 32 bit windows (at least for me). I still get the `exit status 0xc0000005`.

This issue was originally opened for an issue we fixed about a year ago:
- https://github.com/saltosystems/winrt-go/issues/63
- https://github.com/saltosystems/winrt-go/pull/64

I our case the exception seems to be happening on a Windows DLL 😕

roffe, 2024-03-13:
> @aykevl I think we are mixing different problems here. Setting `GOGC="off"` does not help with the error we are seeing in 32 bit windows (at least for me). I still get the `exit status 0xc0000005`.
> 
> This issue was originally opened for an issue we fixed about a year ago:
> 
> * [fatal error: unexpected signal during runtime execution saltosystems/winrt-go#63](https://github.com/saltosystems/winrt-go/issues/63)
> * [codegen: remove go pointers from structs allocated in the heap saltosystems/winrt-go#64](https://github.com/saltosystems/winrt-go/pull/64)
> 
> I our case the exception seems to be happening on a Windows DLL 😕 => [saltosystems/winrt-go#83](https://github.com/saltosystems/winrt-go/issues/83)

I have separate issues open. the 32bit is one thing but this error is produced using 64bit