tinygo-org/bluetooth #131: It cannot find more than one characteristic with the same UUID

viric, 2022-12-04:
I have a temperature sensor peripheral that has an environmental sensing service, and there, two temperatures (same UUID).

A call to DiscoverCharacteristics, with either empty or the Temperature UUID, returns only one of the two characteristics.

Is this a bug or something yet to be implemented?

TirelessDev, 2023-02-24:
This seems like a missing capability that could be platform dependent. Can you provide some more information on what OS you are developing for?

aykevl, 2023-05-03:
@viric here is a partial fix for macOS: https://github.com/tinygo-org/bluetooth/pull/161

deadprogram, 2024-01-07:
This was already released in `v0.8.0` s now closing. Thanks everyone!

closed 2024-01-07 by deadprogram

viric, 2024-01-07:
sorry I was not following in detail. Thanks!