tinygo-org/bluetooth #132: LocalName returning empty on Windows

TirelessDev, 2022-12-21:
Hi guys, 

I am having no luck fetching the LocalName from an advertisement while running on Windows 10 (1903). I have no problem getting the manufacture data, device address, and RSSI. I have tried 3 different computers running Windows 10 and had no luck.

To test, I am using the Central example in README.md and this is an example of what is returned.

```
scanning...
found device: FF:37:4A:A2:AA:60 -63
found device: 76:F8:55:B0:87:C0 -59
found device: F0:B3:EC:54:17:E4 -61
found device: 59:33:42:90:36:A4 -50
found device: FF:09:C7:01:14:2B -63
found device: 4B:06:6E:6D:9E:AC -65
found device: FF:37:4A:A2:AA:60 -66
```

I have tried v0.6.0 as well as the latest on dev. 

I have had a bit of a look myself but quickly got out of my depth when it moved to WinRT.

Cheers

rvt, 2022-12-26:
I am seeing the same on a pi4. To give some background... I have a development on a PI4 where BLE works fairly well, I get the LocalName directly after I do a 'scan' (I also removed the var/lib/bluetooth directly, restart) When I want to do a final build I generate a complete image automatically with the build software and the same (as far as I can see..) software.... After I flash this and try it out, LocalName does not work and also Connection to the bluetooth device does not work, can takes up to 20 minutes before connection finally happens... commenting to follow any advices that might follow on this report.

TirelessDev, 2023-02-16:
Interesting, I have not had any issues with the Pi4. I suspect you are seeing a different issue. As far as I can tell I am experiencing a WinRT issue here.

Are you cross-compiling or building it on the device? Might be worth opening a new issue.

rvt, 2023-02-16:
I just got it fixed last week... 
There where 1..3 packages in the PI4 installed that just really messed up my bluetooth experience.

Right now I have been uninstalling the following _bluetooth pi-bluetooth bluez-firmware_ and bluetooth LE works very good. I always get the name and always connects within seconds. I haven't looked into what each package does and if there is really interest, but I can test it if needed..

Testing is a bit of a long process of generating an image and flash that on a RO filesystem, with the annoying thing is that once BLE works the package doesn't seem matter anymore and I have to re-flash for my next test. Not even clearing /var/lib/bluetooth or restarting then helps to repeat the exact same thing..

ansoni-san, 2023-05-06:
@TirelessDev  I'm having the same issue. 

Other windows apps seem to be able to access LocalName.

Did you have any luck in getting it working? Before I dig in deeper 🙂

TirelessDev, 2023-05-07:
I haven't got any further, sorry. I put it to the side and slapped "windows not supported" on my application.

Keen to see what you can dig up though.

ansoni-san, 2023-05-10:
The majority of devices I find only respond with their name during active scanning. I've opened a pull request in `winrt-go` so that we can set scanning mode to active for go-bluetooth on Windows (the other platforms seem to be doing this implicitly).

PR #164 (2023-05-11): windows: scanning should be set to active
Following this change in `winrt-go` here: https://github.com/saltosystems/winrt-go/pull/70

We can now scan in active mode, which is the current (and best) default behaviour on other platforms.

fixes #132
closed 2023-06-04 by TirelessDev