tinygo-org/bluetooth #135: Device characteristics returned out of order on Linux

SamWhited, 2022-12-24:
The documentation for [`DiscoverCharacteristics`] says:

> If there is no error, the characteristics slice has the same length as the UUID slice with characteristics in the same order in the slice as in the requested UUID list. 

However, at least on  Linux, the code appears to just iterate through characteristics appending them to the slice as they are found. The documentation either needs to be changed, or the code updated to reflect the expected behavior.

[`DiscoverCharacteristics`]: https://pkg.go.dev/tinygo.org/x/bluetooth#DeviceService.DiscoverCharacteristics

tomhut, 2023-03-08:
This issue also seems to be present on macOS/Darwin.

aykevl, 2023-05-03:
Here is a fix for macOS: https://github.com/tinygo-org/bluetooth/pull/161
@tomhut can you test this change?

aykevl, 2023-05-04:
And here is a fix for Linux: https://github.com/tinygo-org/bluetooth/pull/162
@SamWhited can you take a look?

SamWhited, 2023-05-05:
Sorry, I don't even remember what I was working on with this or how I was testing it so I don't have a good way to test it anymore :) Thanks for the fix though!

closed 2023-06-11 by deadprogram