tinygo-org/bluetooth #137: `bluetooth: not found` on `xiao-ble`

drbh, 2023-01-10:
Board: [Seeed Studio XIAO nRF52840](https://www.seeedstudio.com/Seeed-XIAO-BLE-nRF52840-p-5201.html#)

Program build and flashed with

```bash
tinygo flash -target=xiao-ble -port /dev/XXXX main.go
```

```go
package main

import (
    "time"

    "tinygo.org/x/bluetooth"
)

var adapter = bluetooth.DefaultAdapter
var deviceName = "DeviceName"

func main() {
    println("Starting...")
    time.Sleep(5 * time.Second) // allow time to connect to serial port

    err := adapter.Enable()
    if err != nil {
        println(err.Error())
    }
    ch := make(chan bluetooth.ScanResult, 1)
    println("Scanning...")
    if err := adapter.Scan(func(adapter *bluetooth.Adapter, device bluetooth.ScanResult) {
        println("Identified", device.LocalName())
        if device.LocalName() == deviceName {
            println("Matches", deviceName)
            err := adapter.StopScan()
            if err != nil {
                println(err.Error())
            }
            ch <- device
        }
    }); err != nil {
        println(err.Error())
    }

    var device *bluetooth.Device

    select {
    case result := <-ch:
        for {
            println("Gonna To Process")
            if true {
                var err error
                device, err = adapter.Connect(result.Address, bluetooth.ConnectionParams{})
                if err != nil {
                    println(err)
                }
                println("Connected To ", result.Address.String())
                println("Discovering services/characteristics")
                srvcs, err := device.DiscoverServices(nil)
                if err != nil {
                    println(err)
                }
                if len(srvcs) == 0 {
                    println("No Services Found")
                }
                println("Found", len(srvcs), "Services")
                for ser := range srvcs {
                    println("Found Service", srvcs[ser].UUID().String())
                }
            }
            println("Reconnecting in 3 seconds")
            time.Sleep(3 * time.Second)
        }
    }
}
```

Then dump the serial 
```bash
screen /dev/XXXX 9600
```

and I get the following

```
Scanning...
Identified
Identified
Identified DeviceName
Matches DeviceName
Gonna To Process
Connected To  00:00:00:00:00:00
Discovering services/characteristics
bluetooth: not found                      <--- !? why?
No Services Found
Found 0 Services
Reconnecting in 3 seconds
```

It appears that `bluetooth: not found` is the `err` returned from `DiscoverServices`.. I've tried passing `nil` and `[]bluetooth.UUID{}` as arguments and both return the same error.

Is there possibly a library that I'm missing?

deadprogram, 2023-06-06:
@drbh I have this exact board and it is working for me as expected with the latest `dev` branch:

```
$ tinygo flash -target xiao-ble -monitor ./examples/scanner/
Connected to /dev/ttyACM0. Press Ctrl-C to exit.
found device: 47:5D:B9:3F:53:2C -80 
found device: 47:5D:B9:3F:53:2C -75 
found device: 47:5D:B9:3F:53:2C -75 
found device: 47:5D:B9:3F:53:2C -75 
found device: 47:5D:B9:3F:53:2C -75 
found device: 47:5D:B9:3F:53:2C -80
```

Perhaps you can try with the same setup?

deadprogram, 2025-01-09:
I am using a couple of these boards right now for FindMy beacons and they are working well.

@drbh are you still having a problem?