tinygo-org/bluetooth #140: adapter.Scan fatal error: sync: inconsistent mutex state

sercanarga, 2023-02-06:
The "adapter.Scan" function in the [example](https://github.com/tinygo-org/bluetooth/blob/release/examples/scanner/main.go) throws the error "fatal error: sync: inconsistent mutex state" after a certain runtime.

<details>
<summary>Console output</summary>

```bash
fatal error: sync: inconsistent mutex state

goroutine 17 [running, locked to thread]:
runtime.throw({0x7ff76059e03e?, 0x1587c289760?})
        C:/Program Files/Go/src/runtime/panic.go:1047 +0x65 fp=0xc00004bd48 sp=0xc00004bd18 pc=0x7ff7604b6305
sync.throw({0x7ff76059e03e?, 0xc00004be00?})
        C:/Program Files/Go/src/runtime/panic.go:1026 +0x1e fp=0xc00004bd68 sp=0xc00004bd48 pc=0x7ff7604de91e
sync.(*Mutex).lockSlow(0xc00009e2a0)
        C:/Program Files/Go/src/sync/mutex.go:180 +0x245 fp=0xc00004bdb8 sp=0xc00004bd68 pc=0x7ff7604f1165
sync.(*Mutex).Lock(...)
        C:/Program Files/Go/src/sync/mutex.go:90
github.com/saltosystems/winrt-go.(*RefCount).Release(0xc00004be28?)
        C:/Users/pc/go/pkg/mod/github.com/saltosystems/winrt-go@v0.0.0-20220826130236-ddc8202da421/refcount.go:26 +0x54 fp=0xc00004be00 sp=0xc00004bdb8 pc=0x7ff76053ef54
github.com/saltosystems/winrt-go/windows/foundation.winrt_TypedEventHandler_Release(0x25d865ff920)
        C:/Users/pc/go/pkg/mod/github.com/saltosystems/winrt-go@v0.0.0-20220826130236-ddc8202da421/windows/foundation/typedeventhandler_exports.go:69 +0x25 fp=0xc00004be20 sp=0xc00004be00 pc=0x7ff76053fea5
_cgoexp_eec08437a0e4_winrt_TypedEventHandler_Release(0xace41ff6f0)
        _cgo_gotypes.go:177 +0x25 fp=0xc00004be38 sp=0xc00004be20 pc=0x7ff7605402e5
runtime.cgocallbackg1(0x7ff7605402c0, 0xc00004bfe0?, 0x0)
        C:/Program Files/Go/src/runtime/cgocall.go:315 +0x2b8 fp=0xc00004bf00 sp=0xc00004be38 pc=0x7ff760485238
runtime.cgocallbackg(0x0?, 0x0?, 0x0?)
        C:/Program Files/Go/src/runtime/cgocall.go:234 +0x105 fp=0xc00004bf90 sp=0xc00004bf00 pc=0x7ff760484ea5
runtime.cgocallbackg(0x7ff7605402c0, 0xace41ff6f0, 0x0)
        <autogenerated>:1 +0x34 fp=0xc00004bfb8 sp=0xc00004bf90 pc=0x7ff7604e59f4
runtime.cgocallback(0x0, 0x0, 0x0)
        C:/Program Files/Go/src/runtime/asm_amd64.s:998 +0xcf fp=0xc00004bfe0 sp=0xc00004bfb8 pc=0x7ff7604e344f
runtime.goexit()
        C:/Program Files/Go/src/runtime/asm_amd64.s:1598 +0x1 fp=0xc00004bfe8 sp=0xc00004bfe0 pc=0x7ff7604e36a1

goroutine 1 [chan receive]:
runtime.gopark(0xc0000bfc98?, 0x7ff760539d57?, 0xb0?, 0xfc?, 0x7ff760541c18?)
        C:/Program Files/Go/src/runtime/proc.go:381 +0xd6 fp=0xc000425c58 sp=0xc000425c38 pc=0x7ff7604b8e96
runtime.chanrecv(0xc000086120, 0x0, 0x1)
        C:/Program Files/Go/src/runtime/chan.go:583 +0x49d fp=0xc000425ce8 sp=0xc000425c58 pc=0x7ff760487b7d
runtime.chanrecv1(0x25d865fa0e0?, 0xc0000880b0?)
        C:/Program Files/Go/src/runtime/chan.go:442 +0x18 fp=0xc000425d10 sp=0xc000425ce8 pc=0x7ff7604876b8
tinygo.org/x/bluetooth.(*Adapter).Scan(0x7ff76055fee0, 0x7ff7605a2cb0)
        C:/Users/pc/go/pkg/mod/tinygo.org/x/bluetooth@v0.6.0/gap_windows.go:92 +0x4e5 fp=0xc000425f60 sp=0xc000425d10 pc=0x7ff760542a25
main.main()
        C:/Users/pc/GolandProjects/bt-lock/main.go:46 +0x3d fp=0xc000425f80 sp=0xc000425f60 pc=0x7ff760548a3d
runtime.main()
        C:/Program Files/Go/src/runtime/proc.go:250 +0x1f7 fp=0xc000425fe0 sp=0xc000425f80 pc=0x7ff7604b8a77
runtime.goexit()
        C:/Program Files/Go/src/runtime/asm_amd64.s:1598 +0x1 fp=0xc000425fe8 sp=0xc000425fe0 pc=0x7ff7604e36a1

goroutine 2 [force gc (idle)]:
runtime.gopark(0x0?, 0x0?, 0x0?, 0x0?, 0x0?)
        C:/Program Files/Go/src/runtime/proc.go:381 +0xd6 fp=0xc000045fb0 sp=0xc000045f90 pc=0x7ff7604b8e96
runtime.goparkunlock(...)
        C:/Program Files/Go/src/runtime/proc.go:387
runtime.forcegchelper()
        C:/Program Files/Go/src/runtime/proc.go:305 +0xb2 fp=0xc000045fe0 sp=0xc000045fb0 pc=0x7ff7604b8cb2
runtime.goexit()
        C:/Program Files/Go/src/runtime/asm_amd64.s:1598 +0x1 fp=0xc000045fe8 sp=0xc000045fe0 pc=0x7ff7604e36a1
created by runtime.init.6
        C:/Program Files/Go/src/runtime/proc.go:293 +0x25

goroutine 3 [GC sweep wait]:
runtime.gopark(0x1?, 0x0?, 0x0?, 0x0?, 0x0?)
        C:/Program Files/Go/src/runtime/proc.go:381 +0xd6 fp=0xc000047f80 sp=0xc000047f60 pc=0x7ff7604b8e96
runtime.goparkunlock(...)
        C:/Program Files/Go/src/runtime/proc.go:387
runtime.bgsweep(0x0?)
        C:/Program Files/Go/src/runtime/mgcsweep.go:319 +0xde fp=0xc000047fc8 sp=0xc000047f80 pc=0x7ff7604a381e
runtime.gcenable.func1()
[... 96 lines cut ...]
goroutine 6 [GC worker (idle)]:
runtime.gopark(0x1563321c00c?, 0x0?, 0x0?, 0x0?, 0x0?)
        C:/Program Files/Go/src/runtime/proc.go:381 +0xd6 fp=0xc000065f50 sp=0xc000065f30 pc=0x7ff7604b8e96
runtime.gcBgMarkWorker()
        C:/Program Files/Go/src/runtime/mgc.go:1275 +0xf1 fp=0xc000065fe0 sp=0xc000065f50 pc=0x7ff76049a7b1
runtime.goexit()
        C:/Program Files/Go/src/runtime/asm_amd64.s:1598 +0x1 fp=0xc000065fe8 sp=0xc000065fe0 pc=0x7ff7604e36a1
created by runtime.gcBgMarkStartWorkers
runtime.gcBgMarkWorker()
        C:/Program Files/Go/src/runtime/mgc.go:1275 +0xf1 fp=0xc000061fe0 sp=0xc000061f50 pc=0x7ff76049a7b1
runtime.goexit()
        C:/Program Files/Go/src/runtime/asm_amd64.s:1598 +0x1 fp=0xc000061fe8 sp=0xc000061fe0 pc=0x7ff7604e36a1
created by runtime.gcBgMarkStartWorkers
        C:/Program Files/Go/src/runtime/mgc.go:1199 +0x25

goroutine 7 [GC worker (idle)]:
runtime.gopark(0x1563321c00c?, 0x0?, 0x0?, 0x0?, 0x0?)
        C:/Program Files/Go/src/runtime/proc.go:381 +0xd6 fp=0xc000067f50 sp=0xc000067f30 pc=0x7ff7604b8e96
runtime.gcBgMarkWorker()
        C:/Program Files/Go/src/runtime/mgc.go:1275 +0xf1 fp=0xc000067fe0 sp=0xc000067f50 pc=0x7ff76049a7b1
runtime.goexit()
        C:/Program Files/Go/src/runtime/asm_amd64.s:1598 +0x1 fp=0xc000067fe8 sp=0xc000067fe0 pc=0x7ff7604e36a1
created by runtime.gcBgMarkStartWorkers
        C:/Program Files/Go/src/runtime/mgc.go:1199 +0x25
```

kpelzel, 2023-04-16:
I'm also getting this error on windows, but mine happens when I call `bluetooth.DeviceCharacteristic.Write()` 

```go
func setColor(dChar *bluetooth.DeviceCharacteristic, red, green, blue byte) error {
	_, err := dChar.Write([]byte{0x56, red, green, blue, 0x00, 0xF0, 0xAA})
	return err
}
```

```
fatal error: sync: inconsistent mutex state

goroutine 19 [running]:
runtime.throw({0x7ff691ee278d?, 0x7c81d511c78?})
        C:/Program Files/Go/src/runtime/panic.go:1047 +0x65 fp=0xc000045d00 sp=0xc000045cd0 pc=0x7ff691c57e45
sync.throw({0x7ff691ee278d?, 0x6?})
        C:/Program Files/Go/src/runtime/panic.go:1026 +0x1e fp=0xc000045d20 sp=0xc000045d00 pc=0x7ff691c817de
sync.(*Mutex).lockSlow(0x194cd7b3658)
        C:/Program Files/Go/src/sync/mutex.go:180 +0x245 fp=0xc000045d70 sp=0xc000045d20 pc=0x7ff691c91945
sync.(*Mutex).Lock(...)
        C:/Program Files/Go/src/sync/mutex.go:90
github.com/saltosystems/winrt-go/windows/foundation.(*AsyncOperationCompletedHandler).addRef(0x7ff691e9f700?)
        C:/Users/LALT/go/pkg/mod/github.com/saltosystems/winrt-go@v0.0.0-20230124093143-967a889c6c8f/windows/foundation/asyncoperationcompletedhandler.go:76 +0x5b fp=0xc000045dc0 sp=0xc000045d70 pc=0x7ff691dd64bb
bluetooth busy, color not sent
github.com/saltosystems/winrt-go/windows/foundation.NewAsyncOperationCompletedHandler(0xc0000c9490, 0xc00009cea0)
        C:/Users/LALT/go/pkg/mod/github.com/saltosystems/winrt-go@v0.0.0-20230124093143-967a889c6c8f/windows/foundation/asyncoperationcompletedhandler.go:70 +0xd1 fp=0xc000045df0 sp=0xc000045dc0 pc=0x7ff691dd6431
tinygo.org/x/bluetooth.awaitAsyncOperation(0x194a5f39190?, {0x7ff691eeb17d, 0x52})
        C:/Users/LALT/go/pkg/mod/tinygo.org/x/bluetooth@v0.6.0/adapter_windows.go:43 +0x10b fp=0xc000045e78 sp=0xc000045df0 pc=0x7ff691ddbc8b
tinygo.org/x/bluetooth.DeviceCharacteristic.write({{0x5f9b34fb, 0x80000080, 0x1000, 0xffd9}, 0x194a5f39190, 0xc}, {0xc000045f81, 0x7, 0x7}, 0x1?)
        C:/Users/LALT/go/pkg/mod/tinygo.org/x/bluetooth@v0.6.0/gattc_windows.go:286 +0x133 fp=0xc000045ef8 sp=0xc000045e78 pc=0x7ff691ddded3
tinygo.org/x/bluetooth.DeviceCharacteristic.WriteWithoutResponse({{0x5f9b34fb, 0x80000080, 0x1000, 0xffd9}, 0x194a5f39190, 0xc}, {0xc000045f81?, 0x0?, 0x0?})
        C:/Users/LALT/go/pkg/mod/tinygo.org/x/bluetooth@v0.6.0/gattc_windows.go:262 +0x5a fp=0xc000045f48 sp=0xc000045ef8 pc=0x7ff691dddd5a```

BobyClaws, 2023-07-07:
i get the same issue

ansoni-san, 2023-07-08:
> i get the same issue

Can you try the dev branch and see if it is fixed?

BobyClaws, 2023-07-08:
> > i get the same issue
> 
> Can you try the dev branch and see if it is fixed?

haven't tested extensively. but suprisingly yes!

ansoni-san, 2023-07-10:
I thought so. There were a few issues before but Windows works very well now 👍

ansoni-san, 2023-07-10:
Yes, this bug was fixed in winrt-go a few months ago. 

https://github.com/saltosystems/winrt-go/issues/71

And another one was fixed also.

BobyClaws, 2023-07-12:
working great for me as well, probably can be closed.

aykevl, 2023-07-12:
Closing, as this issue appears to be fixed.

closed 2023-07-12 by aykevl

SydneyOwl, 2023-08-21:
Hello,
I encountered the same issue in the latest version, but the problem disappeared after switching to the dev branch. Seems like this bug is not officially fixed?