tinygo-org/bluetooth #141: gattc_windows.go: notify vs. indicate

adn77, 2023-02-07:
I would like to get my Timular double-pyramid (orientation device)  to work with something other than the stock software.

I am new to GO but managed to connect and interview the device, thanks to the excellent samples.

It turns out the characteristics property does not have the *notification* flag set but the *indication* flag instead.
Simply commenting out the check in **gattc_windows.go** didn't work and I believe to even understand why.

The handle format for *notification* (HVN) and *indication* (HVI) is the same but for *indication* the server expects an acknowledgement from the client in the form of a confirmation (HVC).
(http://what-when-how.com/Tutorial/topic-505haskm/Getting-Started-with-Bluetooth-Low-Energy-86.html)

Would I be wrong, to hack the *indication* into the **EnableNotifications** from this PR https://github.com/tinygo-org/bluetooth/pull/116/files ?

adn77, 2023-02-18:
@jagobagascon would you be able to assist with this by any chance?

I Imagine **EnableNotifications** could be extended to also handle *indications*
By determining the type of the returned message (*value notification* vs. *value indication*) it should be possible to send a *value confirmation* if necessary.

Unfortunately I have zero clue about Go programming 🤣

jagobagascon, 2023-02-20:
Hi! I would say that the indication acknowledgement is automatically handled by the OS. At least I didn't see anything in the docs: https://learn.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.genericattributeprofile.gattcharacteristic.valuechanged?view=winrt-22621

**But for it to work you need to explicitly activate indications**. So apart from commenting out the check in **gattc_windows.go** did you try changing the `GattClientCharacteristicConfigurationDescriptorValueNotify` for `GattClientCharacteristicConfigurationDescriptorValueIndicate` here?
https://github.com/tinygo-org/bluetooth/blob/e79ea1e4e99c2628e7219de0bd88da59adaf3d60/gattc_windows.go#L396

I think that should do the trick.

I guess we could improve this by automatically picking between notifications or indications based on the attribute properties. And if both are supported then just pick one of them. MacOS for example already does this ([docs](https://developer.apple.com/documentation/corebluetooth/cbperipheral/1518949-setnotifyvalue)):
`If the specified characteristic’s configuration allows both notifications and indications, calling this method enables notifications only`

adn77, 2023-02-20:
Thank you for your help, it worked instantly!
 😄

closed 2023-03-12 by adn77