tinygo-org/bluetooth #144: Implemented Addresser setters have no effect

TirelessDev, 2023-02-16:
The `Addresser` interface declares the `Set(val string)` and `SetRandom(bool)` receivers but all implementations implement these as value receivers that ineffectively attempt to modify their own state. This is triggering the go static check [SA4005](https://staticcheck.io/docs/checks#SA4005).

https://github.com/tinygo-org/bluetooth/blob/e79ea1e4e99c2628e7219de0bd88da59adaf3d60/gap_darwin.go#L27-L33

These implementations should be pointer receivers but changing them will unfortunately modify the API. All `Address` objects would need to be passed around as `*Address` pointers as the `Address` type will no longer implement the `Addresser` interface. This is most obvious in `func (a *Adapter) Connect(address Addresser, params ConnectionParams) (*Device, error)`.

deadprogram, 2023-04-26:
See the recent updates to `dev` which change this in order to avoid heap allocations.

aykevl, 2023-04-27:
This is indeed a bug, thanks for spotting it!
Here is a fix: https://github.com/tinygo-org/bluetooth/pull/158

> These implementations should be pointer receivers but changing them will unfortunately modify the API. All `Address` objects would need to be passed around as `*Address` pointers as the `Address` type will no longer implement the `Addresser` interface.

At least with the current dev branch, no API change is necessary.

TirelessDev, 2023-05-02:
Awesome! Thanks all, the changes look good. I have also updated PR #149 to reflect the latest dev.

closed 2023-06-11 by deadprogram