tinygo-org/bluetooth #145: Problem with SetConnectedHandler() on Linux (Raspbian) Peripheral Device

SVdvr, 2023-02-17:
Hello,

I am using your lib to create/manage a peripheral BLE device on an RPI 4.

It already works quite well, I can advertise and read/write with services and characteristics I defined. However, I would need to run a function each time a a Central device connects/disconnects to my RPI.
As I understood, the 'adapter.SetConnectHandler()' can be used for that, but it doesn't work for me.
Did I miss something ? My code is very similar to this example : https://github.com/tinygo-org/bluetooth/blob/v0.6.0/examples/circuitplay/main.go
```
adapter.SetConnectHandler(func(d` bluetooth.Addresser, c bool) {
		connected = c

		if !connected && !disconnected {
			clearLEDS()
			disconnected = true
		}

		if connected {
			disconnected = false
		}
	})

	must("enable BLE stack", adapter.Enable())
	adv := adapter.DefaultAdvertisement()
	must("config adv", adv.Configure(bluetooth.AdvertisementOptions{
		LocalName: "TinyGo colors",
	}))
	must("start adv", adv.Start())
```

How do I (is it possible to ?) configure a handler that will be called each time a central device connects to my Linux (Raspbian) peripheral ?

Thank you !

TirelessDev, 2023-02-21:
`SetConnectedHandler()` or more accurately `adapter.connectHandler`  is called when the adaptor is acting as a central itself and a connection is established to a peripheral. It is not called on disconnect in Linux, MacOS, or Windows, only for the embedded nordic implementations. However, I am working on a PR that adds this to the MacOS and Linux implementations, you can find it in my fork if you are interested.

For your use case, you need to implement the callbacks for the peripheral services that you are implementing. I am not aware of a call to `adapter.connectHandler` for any peripheral connection events or a similar alternative. However, I have not used the peripheral side of this library so have likely missed something.

At the bottom of the go library abstractions the BlueZ stack is used for the Linux implementation. You can find its docs here 
[git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc). This is wrapped by [github.com/muka/go-bluetooth](https://github.com/muka/go-bluetooth) which tinygo uses, so by looking through there you can see if the functionality you are looking for is possible and work backwards into the tinygo library to see if it is exposed, or how it may be implemented.

deadprogram, 2023-02-21:
What @TirelessDev said! :smile_cat:

SVdvr, 2023-02-24:
Hello,
I used [github.com/muka/go-bluetooth](https://github.com/muka/go-bluetooth) to monitor on DBus Property "Connected" (I guess that's what @TirelessDev did on his fork, but I saw it after I did it myself 😅) and I managed to do what I wanted ( call a function when a central device connects / disconnects to my peripheral )

I guess I will modify it later to do like @TirelessDev, as it seems much more elegant that what I did ( https://github.com/tinygo-org/bluetooth/pull/149/commits/80d9f9dca455c7750246e50cb2ccc9fdf97fcefe )

Anyway, thank you for your help, it works well !

TirelessDev, 2023-02-24:
Nice, good to know that that property change is fired off when acting as a peripheral as well! 

It might then be worth weaving the functionality into this library in the same way, though I am not sure if the same would be possible on MacOS and Windows. @SVdvr did you want to put a draft PR together with what you have working so that we can look at doing this?

SVdvr, 2023-03-01:
Hello, sorry for the delay, I'm still working on it. 
@TirelessDev I never did PRs before, so I don't really know how it works . Anyway I'll try to explain you what I did, and let you decide if we should make a PR with it.

My first solution was to watch on dbus myself (adding match rules, capturing InterfacesAdded and PropertyChanged messages, unwrapping and processing them), but then I found that  [github.com/muka/go-bluetooth](https://github.com/muka/go-bluetooth) already offered such functions (presented below), so I decided to use them, and the code is easier to understand.

First of all, I wanted to keep an eye on all the currently registered devices, so I created a map containing Device1 objects and PropertyChanged channels: 

```
type DeviceAndPropChangedChannel struct {
	dev      *device.Device1
	chanProp chan *bluez.PropertyChanged
}
var(
        managed_devices = make(map[string]DeviceAndPropChannel)
)
```
Then, I created a goroutine that manages a device's connection state (looks like what you did in your fork) : 

```
// Watches on a device's 'Connected' property
func ManageDeviceConnection(d *device.Device1) {
	chanPC, err := d.WatchProperties()
	managed_devices[string(d.Path())] = DeviceAndPropChangedChannel{d, chanPC}
	CheckErr_F(err, "WatchProperties")
	for prop := range chanPC {
		if prop != nil {
			switch prop.Name {
			case "Connected":
				setConnected(d)
			}
		}
	}
	println("Device", d.Properties.Name, "@", d.Properties.Address, "removed")
}
```

And then, while intializing bluetooth: 

1. Run the goroutine with already discovered devices : 
```
// Create goroutines that watch on the 'Connected' status of every known device
	devs, err := app.Adapter().GetDevices()
	CheckErr_F(err, "GetSavedDevices")
	for _, dev := range devs {
		go ManageDeviceConnection(dev)
	}
```
2. Run a goroutine to create new Device1 when an 'InterfaceAdded' is called on DBus and add it to the map, or close its PropertyChanged channel and remove it from the map when an 'InterfaceRemoved' is called on DBus
```
//Create a new device when an 'InterfaceAdded' is detected on DBus
	chanDD, _, err := app.Adapter().OnDeviceDiscovered()
	CheckErr_F(err, "OnDeviceDiscovered")
	go manageDeviceDiscovery(chanDD)
```

```
// Watches on 'Device Discovered' channel and creates a new device to manage when it happens
func manageDeviceDiscovery(chanDD chan *adapter.DeviceDiscovered) {
	for {
		val := <-chanDD
		if val != nil {
			if val.Type == adapter.DeviceAdded {
				debugln("Device added")
				d, _ := device.NewDevice1(val.Path)
				//CheckErr_L(err, "NewDevice1")
				if d != nil {
					setConnected(d)
					go ManageDeviceConnection(d)
				}
			} else {
				debugln("Device removed")
				if dProp, exists := managed_devices[string(val.Path)]; exists {
					d := dProp.dev
					channel := dProp.chanProp
					d.UnwatchProperties(channel)
					delete(managed_devices, string(val.Path))
				}
			}

		}
	}
}
```
As I only used [github.com/muka/go-bluetooth](https://github.com/muka/go-bluetooth) and not your fork, some of my code is probably a duplicate of yours, sorry about that. If you want I could try to add it to your fork (once again I never did it before, so I don't know how long it would take)

If you see something I could improve in my code, please let me know, I'm also a beginner in Go...

Thank you, have a nice day !

deadprogram, 2023-05-01:
Yes please!

SVdvr, 2023-05-03:
Hello, sorry for the delay, I completely forgot to answer the last post...
I can try to add this functionnality on TirelessDev's fork, but it will work only for Linux, as I only used https://github.com/muka/go-bluetooth .

I'll try to do it during the nex days/weeks, but I'll still tell you how it works, maybe someone will find a better idea.
As a peripheral : 
1. When initializing, get all the bonded devices (there's a function for that in https://github.com/muka/go-bluetooth) and instanciate them
2. For each device, use _d.device.WatchProperties()_ to watch DBus's "Property Changed" channel (as TirelessDev did for central devices on is fork). With this, we can detect when a device connects/disconnects/pairs/bonds.
3. Since we don't trigger a connection ourselves (unlike central devices), we also have to watch DBus's "InterfaceAdded" channel, so that we can instanciate a new device object when a central connects to our peripheral.

deadprogram, 2025-01-30:
Now closing as part of release `v0.11.0`. Thank you!

closed 2025-01-30 by deadprogram

kevin-activ, 2025-04-07:
Unfortunately this is still not working with v0.11.0 on Raspberry Pi Linux

I'm running this lib version:
` Dep: &{Path:tinygo.org/x/bluetooth Version:v0.11.0 Sum:h1:32ludjNnqz6RyVRpmw2qgod7NvDePbBTWXkJm6jj4cg= Replace:<nil>}`

and the latest version of Pi OS
`Linux pi4work 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux`

I have my program running as a BLE peripheral and the connections work just fine and I can send data, but the connect handler is never called. No matter where I call SetConnectHandler() from. I've tried before and after enabling the adapter, before and after starting the advertising, etc 
```
adapter.SetConnectHandler(func(d bluetooth.Device, c bool) {
		println("handler")
		connected = c
	})
```