tinygo-org/bluetooth #150: [Help need] LocalName scanned different from the peripheral set on linux

hainesc, 2023-02-22:
I was running heartrate examples. And as you can see https://github.com/tinygo-org/bluetooth/blob/release/examples/heartrate/main.go#L20, the heartrate peripheral set its localname to 'Go HRS'.

But, when I run heartrate-monitor, I never saw any peripheral with LocalName() == "Go HRS", only line like this:
`found device: 00:1A:7D:DA:71:11 0 demo #1`

Yes, this was the bluetooth device that I run heartrate and I can get heartbeat data, But I don't know why `demo #1` instead of 'Go HRS' here. Can someone tell me why? Thanks.

slntopp, 2023-02-22:
I have simillar issue. Running with `LocalName` like "My Prototype" from RaspberryPI with hostname `pi-proto`

1. When scanning Bluetooth devices from iPhone device is displayed as `pi-proto`
2. When scanning using Gnome bluetooth frontend, discovered name is "jumping" between intended(My Prototype) and `pi-proto`

TirelessDev, 2023-02-24:
Assuming the scanning is performed on Linux, this could then be a caching issue. Perhaps try clearing the `/var/lib/blueooth` folder and trying again?

rvt, 2023-02-24:
On a pi I noticed that had to remove a few modules to prevent the '00:1A:7D:DA:71:11' instead of the name.

I think it's one of these 3 modules that caused me issues `bluetooth pi-bluetooth bluez-firmware`

If you see different name instead of the MAC, it might be indeed caching..

hainesc, 2023-02-27:
@rvt Thanks, it works when I clean the cache.

closed 2023-02-27 by hainesc