tinygo-org/bluetooth #151: undefined: AsyncOperationCompletedHandler when running examples in Windows 10

GabeJerome, 2023-03-09:
I tried to run several different examples on Windows 10, however got the following error output:

```
# github.com/saltosystems/winrt-go/windows/foundation
..\..\go\pkg\mod\github.com\saltosystems\winrt-go@v0.0.0-20220826130236-ddc8202da421\windows\foundation\iasyncoperation.go:34:49: undefined: AsyncOperationCompletedHandler
..\..\go\pkg\mod\github.com\saltosystems\winrt-go@v0.0.0-20220826130236-ddc8202da421\windows\foundation\iasyncoperation.go:48:44: undefined: AsyncOperationCompletedHandler
..\..\go\pkg\mod\github.com\saltosystems\winrt-go@v0.0.0-20220826130236-ddc8202da421\windows\foundation\iasyncoperation.go:49:11: undefined: AsyncOperationCompletedHandler
```

I have the latest version of Go installed.

tomhut, 2023-03-15:
I'm running into the same issue on Windows 11, I've installed mingw-w64 however I'm still seeing the same error when trying to run my app. Any pointers would be greatly appreciated.

tomhut, 2023-03-15:
@GabeJerome I managed to get this working! Here is what I did
1) Install MSYS2 from here https://www.msys2.org/

2) Once installed, open the MSYS2 app and run the following command to install mingw
`pacman -S mingw-w64-x86_64-toolchain mingw-w64-x86_64-pkg-config`

3) Open a command prompt and add a path variable to mingw64 with the following command
`setx path “%PATH%;C:\msys64\mingw64\bin”`

4) Restart command prompt

5) Navigate to your go directory in command prompt and ensure CGO is enabled by running the following command
`go env -w CGO_ENABLED=1`

6) You may also need to run `go generate` in your project directory, after this everything worked for me.

GabeJerome, 2023-04-06:
@tomhut Thanks! this solved my problem right away!

closed 2023-04-06 by GabeJerome