tinygo-org/bluetooth #153: Provide `Write` capability on `DeviceCharacteristic`

BYEDUCK, 2023-04-05:
Is it possible for this library to provide `Write` method on `DeviceCharacteristic` ?
Some devices have disabled `writeWithoutResponse` and only enable `write` on their characteristics.

BYEDUCK, 2023-04-05:
Isn't

```go
func (c DeviceCharacteristic) Write(p []byte) (n int, err error) {
	c.service.device.prph.WriteCharacteristic(p, c.characteristic, true)

	return len(p), nil
}
```

in `gattc_darwin.go` all that is needed ?

aykevl, 2023-04-27:
I would suggest `WriteRequest` instead of `Write`.
If you want to make a pull request, please implement this for at least two different BLE stacks, and test it on both stacks. This makes sure the API is portable.

For example, here is an implementation for Linux I wrote a few days ago:

```diff
diff --git a/gattc_linux.go b/gattc_linux.go
index 4d3fcb5..7a5ec47 100644
--- a/gattc_linux.go
+++ b/gattc_linux.go
@@ -207,12 +207,20 @@ func (s *DeviceService) DiscoverCharacteristics(uuids []UUID) ([]DeviceCharacter
        return chars, nil
 }
 
+func (c DeviceCharacteristic) WriteRequest(p []byte) (n int, err error) {
+       err = c.characteristic.WriteValue(p, map[string]interface{}{"type": "request"})
+       if err != nil {
+               return 0, err
+       }
+       return len(p), nil
+}
+
 // WriteWithoutResponse replaces the characteristic value with a new value. The
 // call will return before all data has been written. A limited number of such
 // writes can be in flight at any given time. This call is also known as a
 // "write command" (as opposed to a write request).
 func (c DeviceCharacteristic) WriteWithoutResponse(p []byte) (n int, err error) {
-       err = c.characteristic.WriteValue(p, nil)
+       err = c.characteristic.WriteValue(p, map[string]interface{}{"type": "command"})
        if err != nil {
                return 0, err
        }
```

ansoni-san, 2023-05-10:
@aykevi I believe `gattc_windows.go` already uses `Write` to provide write with response.

Should that be changed as well? It may break compatibility unnecessarily.

aykevl, 2023-05-10:
> @aykevi I believe `gattc_windows.go` already uses `Write` to provide write with response.

Hmm, yes, you are right. `Write` it is, then.

(Also, I realized there are two different ways of calling them, either "write"/"write without response" or "write request"/"write command" - perhaps it's best to not mix those two).

BYEDUCK, 2023-05-10:
Yes, these are two separate concepts:

`write` === `write request`

`write without response` === `write command`

`gate_windows.go` indeed already has `write` capability

I've tested the code with changes I mentioned and everything seemed fine at the beginning but with two different sequential write requests when I read response for the second one I still got response for the first.

I didn't have time to investigate what is the problem.

I switched to [bleak](https://github.com/hbldh/bleak) and can recommend it if you are not sticked to golang.

Ziphone, 2023-11-10:
The suggestion of @BYEDUCK did the trick for me. Did an awful workaround using `go mod vendor` - rather than maintaining a fork. Hope to see `Write` for Darwin in the future.

acouvreur, 2026-10-17:
Fixed by #212 (darwin) and #389 (linux).

closed 2026-10-17 by acouvreur