tinygo-org/bluetooth #154: Access RSSI after connection has been made

GabeJerome, 2023-04-06:
I am connected to my device and pulling data from it with no problem. However, I want to display the current signal strength for the device, but ScanResult.RSSI only contains the RSSI from when a packet was last received from the device (when it was scanned). **Is there a way I can continuously obtain the RSSI after the connection has been made?** I have attempted to set up a watcher to wait for advertisements but that will not work because I am already connected to the device and I want to stay connected to it.

aykevl, 2023-04-06:
For which BLE stack is this? (Linux, MacOS, Windows, Nordic SoftDevice)

GabeJerome, 2023-04-06:
@aykevl This is on Windows 10.

aykevl, 2023-05-01:
This is not currently supported in the Bluetooth package, and I don't know how it works on Windows.
On the Nordic SoftDevice there is the `sd_ble_gap_rssi_get` function for this purpose.