tinygo-org/bluetooth #155: Avoid allocations in interrupt handlers

ysoldak, 2023-04-16:
My app for NRF52840 panics with following message when compiled with latest TinyGo.

`panic: runtime error: alloc in interrupt`

Previously, up to TinyGo `v0.25.0` it was just working.
Starting with TinyGo `v0.26.0` it fails with `out of memory` panic.
Recently, it begun showing this correct panic message, about allocations in interrupts and I've traced problem down to this lib.

dgryski, 2023-04-16:
Question : was it allocating previously and not warning, or was it not allocating?

ysoldak, 2023-04-16:
> Question : was it allocating previously and not warning, or was it not allocating?

This lib has been allocating always, and still is.
The problem is (at least) here:
https://github.com/tinygo-org/bluetooth/blob/release/adapter_sd.go#L72-L88

Also, many allocations in `handleEvent()` function.

aykevl, 2023-04-19:
It's probably this heap allocation here:
https://github.com/tinygo-org/bluetooth/blob/e79ea1e4e99c2628e7219de0bd88da59adaf3d60/adapter_nrf528xx-full.go#L80-L83
`globalScanResult.Address` is an Addresser, which is an interface, so those lines convert the concrete type `Address` into an interface type, which indeed results in a heap allocation.

I've known about this issue for a while now (see https://github.com/tinygo-org/tinygo/issues/1461#issuecomment-826169142) but never got around to fixing it.

(I have an idea for how to log source locations for panics in `-monitor` with very little overhead, maybe I can implement it some day to make debugging these issues easier. The basic idea is to print the PC register, detect the pattern in the monitor, and look up the source location for the given PC address in the ELF file).

aykevl, 2023-04-19:
> Question : was it allocating previously and not warning, or was it not allocating?

It was previously allocating but not warning, leading to potential memory corruption if it happened at the wrong moment. See the discussion in https://github.com/tinygo-org/tinygo/pull/1460.

ysoldak, 2023-04-19:
Ah, okay, I misunderstood the problem then.
Only Heap allocations are bad in context of interrupt handlers, local allocations are fine, of course.

Suggestion: update the panic message to explicitly say "heap" in it: `panic: runtime error: heap alloc in interrupt`

ysoldak, 2023-04-25:
Works for me now with `fix-panics` branch and after eliminating all heap allocations in user code in `SetConnectHandler()`

closed 2023-04-27 by ysoldak