tinygo-org/bluetooth #165: How should I send about two dozen attribute descriptors via UUID? I would like to emulate my Linux Bluetooth module as a HID device.

nnnpa31, 2023-05-10:
I just have an XML file like this:

```xml
<?xml version="1.0" encoding="UTF-8" ?>
<record>
    <!-- service class id list -->
    <attribute id="0x0001">
        <sequence>
            <!-- HID -->
            <uuid value="0x1124" />
        </sequence>
    </attribute>
    <!-- protocol descriptor list -->
    <attribute id="0x0004">
        <sequence>
            <sequence>
                <!-- L2CAP -->
                <uuid value="0x0100" />
                <!-- HID Control PSM -->
                <uint16 value="0x0011" />
            </sequence>
            <sequence>
                <!-- HID -->
                <uuid value="0x0011" />
            </sequence>
        </sequence>
    </attribute>
    <!-- language base attribute ID list -->
    <attribute id="0x0006">
        <sequence>
            <uint16 value="0x656e" />
            <uint16 value="0x006a" />
            <uint16 value="0x0100" />
        </sequence>
    </attribute>
    <!-- profile descriptor list -->
    <attribute id="0x0009">
        <sequence>
            <sequence>
                <!-- UUID -->
                <uuid value="0x1124" />
                <!-- version -->
                <uint16 value="0x0100" />
            </sequence>
        </sequence>
    </attribute>
    <!-- additional protocol descriptor list -->
    <attribute id="0x000d">
        <sequence>
            <sequence>
                <!-- L2CAP PSM -->
                <sequence>
                    <uuid value="0x0100" />
                    <uint16 value="0x0013" />
                </sequence>
                <!-- HID -->
                <sequence>
                    <uuid value="0x0011" />
                </sequence>
            </sequence>
        </sequence>
    </attribute>
    <!-- service name -->
    <attribute id="0x0100">
        <text value="Virtual Keyboard" />
    </attribute>
    <!-- service description -->
    <attribute id="0x0101">
        <text value="BT Keyboard" />
    </attribute>
    <!-- service provider name -->
    <attribute id="0x0102">
        <text value="Lv Haitao" />
    </attribute>
    <!-- HIDParserVersion -->
    <attribute id="0x0201">
        <uint16 value="0x0111" />
    </attribute>
    <!-- HIDDeviceSubclass -->
    <attribute id="0x0202">
        <uint8 value="0x80" />
    </attribute>
    <!-- HIDCountryCode -->
    <attribute id="0x0203">
        <uint8 value="0x00" />
    </attribute>
    <!-- HIDVirtualCable -->
    <attribute id="0x0204">
        <boolean value="true" />
    </attribute>
    <!-- HIDReconnectInitiate -->
    <attribute id="0x0205">
        <boolean value="true" />
    </attribute>
    <!-- HIDDescriptorList -->
    <attribute id="0x0206">
        <sequence>
            <sequence>
                <!-- USB Report -->
                <uint8 value="0x22" />
                <!-- HID Descriptor that is defined in Section 6.2 of the USB HID Specification -->
                <text encoding="hex" value="05010902a10185010901a1000509190129031425017501950381027505950181010501093009311581257f750895028106093895018106c0c00906a101850275019508050719e029e7142501810295017508810395057501050819012905910295017503910395067508256505071829658100c0" />
            </sequence>
        </sequence>
    </attribute>
    <!-- HIDLANGIDBaseList -->
    <attribute id="0x0207">
        <sequence>
            <sequence>
                <uint16 value="0x0409" />
                <uint16 value="0x0100" />
            </sequence>
        </sequence>
    </attribute>
    <!-- HIDSupervisionTimeout -->
    <attribute id="0x020c">
        <uint16 value="0x0c80" />
    </attribute>
    <!-- HIDNormallyConnectable -->
    <attribute id="0x020d">
        <boolean value="false" />
    </attribute>
    <!-- HIDBootDevice -->
    <attribute id="0x020e">
        <boolean value="true" />
    </attribute>
    <!-- HIDSSRHostMaxLatency -->
    <attribute id="0x020f">
        <uint16 value="0x0640" />
    </attribute>
    <!-- HIDSSRHostMinTimeout -->
    <attribute id="0x0210">
        <uint16 value="0x0320" />
    </attribute>
</record>
```

Thank you so much for your willingness to teach me.

aykevl, 2023-05-10:
We currently don't support attribute descriptors. You are free to add support for them, of course.

nnnpa31, 2023-05-11:
TBH, I'm so new at this that I can't use c to solve this problem via BlueZ. But thanks for your reply.
Bluetooth development may be an important part of my future work, and I will come back and contribute when I may have solved this problem.

closed 2023-05-11 by nnnpa31