tinygo-org/bluetooth #175: iMac 27inch (intel i5): error: function does not return string type NS_FORMAT_ARGUMENT

pronvis, 2023-07-15:
Cant execute simple example on iMac with i5 proocessor :(
```
~/go ➜ xcode-select --install
xcode-select: error: command line tools are already installed, use "Software Update" in System Settings to install updates

~/go ➜ git clone https://github.com/tinygo-org/bluetooth.git
Cloning into 'bluetooth'...
remote: Enumerating objects: 1166, done.
remote: Counting objects: 100% (394/394), done.
remote: Compressing objects: 100% (146/146), done.
remote: Total 1166 (delta 277), reused 261 (delta 247), pack-reused 772
Receiving objects: 100% (1166/1166), 1.42 MiB | 4.99 MiB/s, done.
Resolving deltas: 100% (729/729), done.
~/go ➜ cd bluetooth

~/go/bluetooth ➜ go run ./examples/scanner
# github.com/tinygo-org/cbgo
In file included from ../pkg/mod/github.com/tinygo-org/cbgo@v0.0.4/att.go:5:
In file included from ./bt.h:4:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Foundation.framework/Headers/Foundation.h:12:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Foundation.framework/Headers/NSBundle.h:6:
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Foundation.framework/Headers/NSString.h:371:81: error: function does not return string type
- (nullable instancetype)initWithUTF8String:(const char *)nullTerminatedCString NS_FORMAT_ARGUMENT(1);
                                                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^                  ~
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Foundation.framework/Headers/NSObjCRuntime.h:103:48: note: expanded from macro 'NS_FORMAT_ARGUMENT'
        #define NS_FORMAT_ARGUMENT(A) __attribute__ ((format_arg(A)))
                                                      ^          ~
In file included from ../pkg/mod/github.com/tinygo-org/cbgo@v0.0.4/att.go:5:
In file included from ./bt.h:4:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Foundation.framework/Headers/Foundation.h:12:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Foundation.framework/Headers/NSBundle.h:6:
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Foundation.framework/Headers/NSString.h:372:52: error: function does not return NSString
- (instancetype)initWithString:(NSString *)aString NS_FORMAT_ARGUMENT(1);
                                ~~~~~~~~~~~~~~~~~~ ^                  ~
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Foundation.framework/Headers/NSObjCRuntime.h:103:48: note: expanded from macro 'NS_FORMAT_ARGUMENT'
        #define NS_FORMAT_ARGUMENT(A) __attribute__ ((format_arg(A)))
                                                      ^          ~
In file included from ../pkg/mod/github.com/tinygo-org/cbgo@v0.0.4/att.go:5:
In file included from ./bt.h:4:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Foundation.framework/Headers/Foundation.h:12:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Foundation.framework/Headers/NSBundle.h:6:
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Foundation.framework/Headers/NSString.h:387:53: error: function does not return NSString
+ (instancetype)stringWithString:(NSString *)string NS_FORMAT_ARGUMENT(1);
                                  ~~~~~~~~~~~~~~~~~ ^                  ~
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Foundation.framework/Headers/NSObjCRuntime.h:103:48: note: expanded from macro 'NS_FORMAT_ARGUMENT'
        #define NS_FORMAT_ARGUMENT(A) __attribute__ ((format_arg(A)))
                                                      ^          ~
In file included from ../pkg/mod/github.com/tinygo-org/cbgo@v0.0.4/att.go:5:
In file included from ./bt.h:4:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Foundation.framework/Headers/Foundation.h:12:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Foundation.framework/Headers/NSBundle.h:6:
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Foundation.framework/Headers/NSString.h:389:83: error: function does not return string type
+ (nullable instancetype)stringWithUTF8String:(const char *)nullTerminatedCString NS_FORMAT_ARGUMENT(1);
                                                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^                  ~
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Foundation.framework/Headers/NSObjCRuntime.h:103:48: note: expanded from macro 'NS_FORMAT_ARGUMENT'
        #define NS_FORMAT_ARGUMENT(A) __attribute__ ((format_arg(A)))
                                                      ^          ~
In file included from ../pkg/mod/github.com/tinygo-org/cbgo@v0.0.4/att.go:5:
In file included from ./bt.h:4:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Foundation.framework/Headers/Foundation.h:12:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Foundation.framework/Headers/NSBundle.h:6:
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Foundation.framework/Headers/NSString.h:395:114: error: function does not return string type
- (nullable instancetype)initWithCString:(const char *)nullTerminatedCString encoding:(NSStringEncoding)encoding NS_FORMAT_ARGUMENT(1);
                                                ~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                     ^                  ~
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Foundation.framework/Headers/NSObjCRuntime.h:103:48: note: expanded from macro 'NS_FORMAT_ARGUMENT'
        #define NS_FORMAT_ARGUMENT(A) __attribute__ ((format_arg(A)))
                                                      ^          ~
In file included from ../pkg/mod/github.com/tinygo-org/cbgo@v0.0.4/att.go:5:
In file included from ./bt.h:4:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Foundation.framework/Headers/Foundation.h:12:
[... 2 lines cut ...]
+ (nullable instancetype)stringWithCString:(const char *)cString encoding:(NSStringEncoding)enc NS_FORMAT_ARGUMENT(1);
                                                  ~~~~~~~~~~~~~~                                ^                  ~
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Foundation.framework/Headers/NSObjCRuntime.h:103:48: note: expanded from macro 'NS_FORMAT_ARGUMENT'
        #define NS_FORMAT_ARGUMENT(A) __attribute__ ((format_arg(A)))
                                                      ^          ~
In file included from ../pkg/mod/github.com/tinygo-org/cbgo@v0.0.4/att.go:5:
In file included from ./bt.h:4:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Foundation.framework/Headers/Foundation.h:12:
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Foundation.framework/Headers/NSBundle.h:91:143: error: function does not return NSString
- (NSAttributedString *)localizedAttributedStringForKey:(NSString *)key value:(nullable NSString *)value table:(nullable NSString *)tableName NS_FORMAT_ARGUMENT(1) NS_REFINED_FOR_SWIFT API_AVAILABLE(macos(12.0), ios(15.0), watchos(8.0), tvos(15.0));
                                                         ~~~~~~~~~~~~~~                                                                       ^                  ~
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Foundation.framework/Headers/NSObjCRuntime.h:103:48: note: expanded from macro 'NS_FORMAT_ARGUMENT'
        #define NS_FORMAT_ARGUMENT(A) __attribute__ ((format_arg(A)))
                                                      ^          ~
In file included from ../pkg/mod/github.com/tinygo-org/cbgo@v0.0.4/att.go:5:
In file included from ./bt.h:4:
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Foundation.framework/Headers/Foundation.h:130:
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Foundation.framework/Headers/NSURLSession.h:505:186: error: expected ')'
[... cut ...]
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Foundation.framework/Headers/NSURLSession.h:505:159: note: to match this '('
- (void)readDataOfMinLength:(NSUInteger)minBytes maxLength:(NSUInteger)maxBytes timeout:(NSTimeInterval)timeout completionHandler:(void (NS_SWIFT_SENDABLE ^) (NSData * _Nullable_result data, BOOL atEOF, NSError * _Nullable error))completionHandler;
                                                                                                                                                              ^
8 errors generated.
```

aykevl, 2023-07-15:
Hmm, I have no idea what's going wrong there. It certainly is working on recent Macs (it is tested in CI).

To be honest this looks a lot like a bug in the SDK, or a version mismatch, but doesn't look like a bug in the Bluetooth package. From my experience the MacOS SDK often has weird problems like this. Perhaps you can try uninstalling the SDK and reinstalling it?

ansoni-san, 2023-07-15:
What version of macOS are you on? And what version of Xcode?

pronvis, 2023-07-16:
@ansoni-san 

MacOS version: 13.4 
xcode: ```
xcode-select -v
xcode-select version 2397.```

@aykevl

"Perhaps you can try uninstalling the SDK and reinstalling it?" 
Did you mean reinstall  `xcode-select` or Xcode itself? (or is it the same?)

ansoni-san, 2023-07-17:
Try installing Xcode 13.4, and then use xcode-select to switch to that SDK. 

That version of command line tools doesn't seem to match the build target correctly.

pronvis, 2023-07-19:
My Xcode is 14.3.1 version. I dont have another one.
The path is:
```
> xcode-select -p
/Applications/Xcode.app/Contents/Developer
``` 

Which is the same as in Xcode:
[image]

ansoni-san, 2023-07-19:
Yes the problem is your Xcode. It's too new and broken. You'll need a lower version I think 🤔 

Or it may be fixed by changing your build to target the right minimum version of macOS based on whatever Apple has deprecated. In my experience this is usually caused by deprecations or API changes and your build targeting a particular minimum version of macOS. 

These kinds of macros are usually controlled by macOS target version switches. 

Honestly I would just install a lower version of Xcode that has good support everywhere, and be done with it. You should be able to install multiple versions side-by-side and switch between them as necessary (using xcode-select). 🙂

pronvis, 2023-07-19:
Ohh... hate xcode :(
Probably I will not continue with that process :)

aykevl, 2023-07-31:
I don't have a lot of experience with macOS and Xcode, but using an older stable version seems like a reasonable things to try.

deadprogram, 2023-09-10:
We will add support for XCode 14 only after GH actions adds support so we can test as part of our builds. Please see https://github.com/github/roadmap/issues/813