tinygo-org/bluetooth #177: Heartrate Example issue

lidiyag, 2023-07-21:
I am new to tinygo-org/bluetooth package. 
I tried to run the heartrate example in the repo.

My setup is as follows:
1. Peripheral device(Ubuntu Laptop)
https://github.com/tinygo-org/bluetooth/blob/release/examples/heartrate/main.go
I ran this code as the peripheral

2. Central device(Mac Laptop)
https://github.com/tinygo-org/bluetooth/blob/release/examples/heartrate-monitor/main.go
I ran this code as my central device

I could identify from logs that the scanning and connection to peripheral was successful. Also the central discovered the service and characteristic successfully.

I could verify the peripheral is writing successfully to the characteristic as well. But unfortunately, I am not able to receive any notifications on the central. 

// Enable notifications for the characteristic

	errEnable := char.EnableNotifications(func(buf []byte) {
		println("data:", uint8(buf[1])))
	})
	if errEnable != nil {
		println("Enable Error", errEnable)
	} else {
		println("Enabling notfcns on char is success")
	}
	
The enable notification was successful, but it never printed the data on central side. 

I also tried reading the characteristic by adding bluetooth.CharacteristicReadPermission on peripheral and the following code in central

        dataRead := make([]byte, 2)
	len, errRead := char.Read(dataRead)
	println("Length of char data read", len)
	println("data:", int(dataRead[1]))
	println("Error on Data Read", errRead.Error())

This mostly errored with a timeout error on read. It gave me data only once. 
Can someone help me in to figure out why this is happening?

ilidiya, 2023-07-23:
@deadprogram @aykevl please help!

deadprogram, 2023-09-10:
Sounds like the same change as https://github.com/tinygo-org/bluetooth/pull/182 might address this?