tinygo-org/bluetooth #180: How should I call the Notify function of a characteristic?

pekuecoder, 2023-08-02:
My platform is Linux, and to simulate a Bluetooth keyboard, I need to call the Notify method of ReportCharacteristic under the HID service to send the typing of letters. 
I found that when setting Characteristic, you can specify the permission for Notify, but there is no corresponding method.

aykevl, 2023-08-30:
I think what you need to do is simply to write to the characteristic. If the other end has notifications enabled, it will receive a notification when the characteristic is written to (so it behaves more like sending a message than updating some property).

pekuecoder, 2023-09-02:
You are right. I discover this matter later, but thank you replay. I will close this issue.

closed 2023-09-02 by pekuecoder