tinygo-org/bluetooth #184: Startup requirements in Linux

xeanhort, 2023-08-07:
Hi, I'm having an issue with a code that uses this library running in Raspberry.

The code works properly if I start it once the system has been turned on for a few minutes. If I run the code with systemctl (`After=bluetooth.target`) it returns 0 BLE devices when scanning, but there is no error returned in any of the library functions.

I'm not sure if I should wait for a different service (other than `bluetooth.target`), but I think the library should be aware that is not "scanning properly" and return an error if it fails.

Thank you

TirelessDev, 2023-08-18:
This is a problem on macOS and presumably Windows as well. Not so much the boot-up race condition, but the ability to check the adaptor status at run time which would provide a means of working around this.

On macOS at least you will receive an error when calling `adapter.Enable()` if the Bluetooth adapter is turned off but on Linux it seems to just continue happily, at least on Raspbian. There is also no way to tell after calling `Enable()` if the adaptor has been turned off or something else funny has gone on causing the adapter to stop scanning.  The capabilities exist on the macOS `cbgo` side to expose notifications about the adaptor state but it's not as clear how this will be done on the Linux side. 

Unfortunately, I don't have the ability to explore this on Windows or bare metal. However, I suspect on the bare metal side this is largely a non-issue. 

@deadprogram what are your thoughts on exposing adapter state updates?

I will do some digging.

ansoni-san, 2023-08-18:
I have a need for this also.

I have hacked around it for the time being as I wasn't sure about the design decisions, and because probably I'm the wrong person as I would immediately want a fully event-based interface for this project. Needs a longer-term understanding of the design decisions behind the current interface.

I forgot that I also really want this 🤔

TirelessDev, 2023-08-18:
A full event-based system would be nice, but given that the interface needs to be efficient for bare metal targets I expect it would be a bit much. My plan is to implement it as a simple callback on adapter state change. This is the same interface approach used for the GAP connect handler and scanning callback.

aykevl, 2023-08-29:
I've found systemd to be rather confusing, and dependencies don't always do what you'd expect. I guess the Bluetooth service isn't fully started (BT adapter powered on and all) before your custom service starts.

Here is something that may or may not help (I'm still confused after reading these descriptions, they seem way more vague than necessary):
https://serverfault.com/questions/812584/in-systemd-whats-the-difference-between-after-and-requires