tinygo-org/bluetooth #198: heartrate example panic

gitchander, 2023-09-27:
I have next problem with standard example of heartrate:

panic: interface conversion: interface {} is map[string]map[string]dbus.Variant, not dbus.ObjectPath

goroutine 1 [running]:
tinygo.org/x/bluetooth.(*Adapter).Scan(0x786c00, 0xc0000bdea8)
	.../pkg/mod/tinygo.org/x/bluetooth@v0.8.0/gap_linux.go:177 +0xeb7

TirelessDev, 2023-10-01:
Can you provide a bit more information, please? 

What are you running this on? 
Are you using the latest release or a branch?
What version of DBus are you running?

Cheers

gitchander, 2023-10-02:
I do it on:
os Ubuntu 22.04.3 LTS

Adapter USB Bluetooth 5.1 USB 2.0 RTL8761B Chip

go.mod file:

go 1.21.0

require tinygo.org/x/bluetooth v0.8.0

require (
	github.com/fatih/structs v1.1.0 // indirect
	github.com/go-ole/go-ole v1.3.0 // indirect
	github.com/godbus/dbus/v5 v5.1.0 // indirect
	github.com/muka/go-bluetooth v0.0.0-20221213043340-85dc80edc4e1 // indirect
	github.com/saltosystems/winrt-go v0.0.0-20230921082907-2ab5b7d431e1 // indirect
	github.com/sirupsen/logrus v1.9.3 // indirect
	github.com/tinygo-org/cbgo v0.0.4 // indirect
	golang.org/x/sys v0.12.0 // indirect
)

go.sum file:

tinygo.org/x/bluetooth v0.7.0 h1:7lU0VrauwccbLvb6AKHf4ZedFlxZIFwwqHWQD2fLWvA=
tinygo.org/x/bluetooth v0.7.0/go.mod h1:2hZPpfPDMR7Vvi6yMvyi83o8gYXeMKKsLfUnvmtBcjM=
tinygo.org/x/bluetooth v0.8.0 h1:WmuRebsODcUUIlGhesyuNRIAEIUCErhKlrZ9K9aimdI=
tinygo.org/x/bluetooth v0.8.0/go.mod h1:cfsVc0/nGo3nzi6+CeQaXb+anNlmEnSABkKsxer8OAE=

TirelessDev, 2023-10-02:
Sweet thanks, what does `bluetoothctl -v` say?

gitchander, 2023-10-02:
bluetoothctl: 5.64

TirelessDev, 2023-10-02:
Okay, very peculiar. Everything checks out with the versions that I can see.  

The panic seems to suggest that the dbus object that has been provided in the `org.freedesktop.DBus.ObjectManager.InterfacesAdded` signal is somehow backwards with the Variant map in index 0 and the object path in index 1. I have no idea how that would happen though.
https://github.com/tinygo-org/bluetooth/blob/d0c7887b81e136df90b732d1162cdb335633aafa/gap_linux.go#L177-L178

maybe try monitor dbus with  `sudo busctl monitor org.bluez | grep -A4 "InterfacesAdded" ` you should get something like the following show up when new devices are found during the scan.

```
  Sender=:1.212  Path=/  Interface=org.freedesktop.DBus.ObjectManager  Member=InterfacesAdded
  UniqueName=:1.212
  MESSAGE "oa{sa{sv}}" {
          OBJECT_PATH "/org/bluez/hci0/dev_XX_XX_XX_XX_XX_XX";
          ARRAY "{sa{sv}}" {
```

You can also manually enable a scan without running any go code with the command `bluetoothctl scan on`.

gitchander, 2023-10-02:
I scanned and connected to BlueNRG-345 (dev kit STEVAL-IDB), discovered services and read characteristics. There was no problem.

gitchander, 2023-10-03:
len(sig.Body) => 1
sig.Body[0] type is "map[string]map[string]dbus.Variant"

map key "org.bluez.GattService1"
map values:
	"Handle": @q 0
	"Primary": true
	"UUID": "0000180d-0000-1000-8000-00805f9b34fb"
	"Characteristics": @ao []

TirelessDev, 2023-10-05:
Not too sure what could be going on then. If you haven't already I would try reinstalling BlueZ or maybe trying a different version.

gitchander, 2023-11-09:
Can I run heartrate server and client on one PC?

deadprogram, 2024-01-04:
Very good chance this is now fixed by #216 which was merged into `dev` branch.

aykevl, 2024-01-05:
> Can I run heartrate server and client on one PC?

No. But you can use a smartphone app, for example "nRF Connect" on Android.
(I'm personally also using a Raspberry Pi for testing such stuff as a 2nd system).

gitchander, 2024-01-05:
If you start the client first and then start the server on the same PC, this will still lead to panic. Maybe this can be fixed.
Thanks for your answer!

closed 2024-01-05 by gitchander