tinygo-org/bluetooth #204: Secure Connections

gitchander, 2023-11-28:
Can I make bonding Secure Connections for service?

deadprogram, 2023-12-11:
At least some of the current platforms can support this, with some work to create the needed wrappers in this package.

SoftDevice:
https://infocenter.nordicsemi.com/topic/com.nordic.infocenter.s132.api.v5.0.0/group___b_l_e___g_a_p___p_e_r_i_p_h___l_e_s_c___m_s_c.html

Linux:
https://github.com/muka/go-bluetooth/blob/master/bluez/profile/device/gen_Device1.go#L782

aykevl, 2023-12-25:
See: https://github.com/tinygo-org/bluetooth/issues/200 and https://github.com/tinygo-org/bluetooth/pull/201

I plan on working on bonding soon, probably starting with #201 (but making sure the API is entirely cross-platform).

funkycode, 2024-11-30:
I know I didn't finish the previous PR on bonding, but back then I also poked secure bonding and was able to get it to connect, but not to reconnect for some reason (was long ago, so do not recall much) maybe it can be useful too, though it is not much there https://github.com/funkycode/bluetooth/tree/dev-lesc

Didn't touch it for a long time as lack of time, but would check if there is anything more I didn't commit in case it can be useful to someone to make a better progress and cleaner code 😅