tinygo-org/bluetooth #206: le-connection-abort-by-local on Raspberry Pi 4

k2sebeom, 2023-12-18:
I am running on Raspberry Pi 4 Model B to connect with BLE devices.

When I run adapter.Connect, it always returns error "le-connection-abort-by-local."

BlueZ version = 5.66

```golang
package main

import (
	"fmt"
	"strings"
	"time"

	"tinygo.org/x/bluetooth"
)

func main() {
	adapter := bluetooth.DefaultAdapter

	if err := adapter.Enable(); err != nil {
		panic(err)
	}

	scanCh := make(chan bluetooth.ScanResult, 1)

	err := adapter.Scan(func(adapter *bluetooth.Adapter, device bluetooth.ScanResult) {
		println("found device:", device.Address.String(), device.RSSI, device.LocalName())
		if strings.Contains(device.LocalName(), "PD") {
			scanCh <- device
			adapter.StopScan()
		}
	})

	device := <- scanCh
	close(scanCh)
	adapter.SetConnectHandler(func(device bluetooth.Address, connected bool) {
		fmt.Printf("Connected to %s with %v\n", device.String(), connected)
	})

	time.Sleep(time.Second * 1)

	_, err = adapter.Connect(device.Address, bluetooth.ConnectionParams{
		ConnectionTimeout: bluetooth.NewDuration(time.Second * 100),
	})

	if err != nil {
		fmt.Printf("Failed to Connect %s\n", err.Error())
	} else {
		fmt.Printf("Connected to %s\n", device.Address)
	}
}
```

aykevl, 2023-12-27:
Have you tried connecting manually through `bluetoothctl`? Does it work there?

mdisibio, 2024-07-13:
Hi, this also happens sporadically on the Raspberry Pi 5.   Sometimes it works and sometimes it doesn't.  I'm wondering if there is some internal state in the BT stack that the library scan/connect pattern isn't always achieving.  Similarly `bluetoothctl` connect does work, but only after doing `scan on` and letting it find the target device.  Not sure what all data is helpful but here is some basic info:

Using `tinygo.org/x/bluetooth v0.10.0`
```
$ bluetoothd -v
5.66
```
```
$ uname -a
Linux raspberrypi 6.6.31+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64 GNU/Linux
```

This is a 2 part question:

I notice a difference in the behavior of `adapter.Connect` between linux and macos (trying to migrate my app from my laptop to the rasppi).   ConnectionTimeout doesn't seem to be honored on linux - it seems to either connect right away or fail, based on if a scan "found" the device already.  This isn't inside the same go app PID, across different executions. Which seems to indicate the state is lower in the BT stack.     On macos no pre-scan is necessary, can connect straight to the device using a hardcoded addr, and it's not instantaneous, can take 30+ seconds but does honor the timeout and works 99+%.

For linux - is the pre-scan required even if I already know the device addr I want to connect to?

Matthias-Hess, 2024-11-24:
Have you found a solution for this nasty issue? I am beeing haunted by it for days and can't figure it out.

ztolley, 2025-02-22:
I found the timeout seems to be ignored on Raspberry pi also, though the same code worked on linux on intel. I'm going to try Ubuntu on raspberry pi to compare but need to get to the bottom of if this is Raspberry Pi's hardware or the OS

paulkre, 2025-03-03:
I'm facing the same issue.

deadprogram, 2025-03-03:
Sounds a bit like the adaptor is unavailable due to already being busy from `bluetoothctl`.

Perhaps you want to try with the latest `dev` branch of this package which has this PR to help detect some of these states: https://github.com/tinygo-org/bluetooth/pull/351

paulkre, 2025-03-04:
Nevermind, using `bluetoothctl` consistently fails with too. Error message:

```
Failed to connect: org.bluez.Error.Failed le-connection-abort-by-local
```

paulkre, 2025-03-04:
I found the source of the problem: It was caused by HostAP which was using the same network interface.

dhairyashah1, 2025-05-29:
What if I need to use wi-fi and bluetooth together? How to tackle the interference issue?

aykevl, 2025-05-30:
@dhairyashah1 that's not an issue with our Bluetooth package. Perhaps you can ask this in a Raspberry Pi forum for example?

cds-sam-rall, 2025-11-10:
> I found the source of the problem: It was caused by HostAP which was using the same network interface.

I am having this same issue, can you elaborate on your solution? @paulkre