tinygo-org/bluetooth #241: advertising data elements missing

dnlwgnd, 2024-02-14:
Hi,
I continue to explore this package.
I am working on a nrf52 device, my goal is to implement the [BTHome protocol](https://bthome.io/format/)
This seems to require me to add the
`0x16 = Service Data - 16-bit UUID`
advertising data element.
From what I found in the codebase, this particular element (and many more) are not possible to configure, i.e. in
```
// AdvertisementOptions configures an advertisement instance. More options may
// be added over time.
type AdvertisementOptions struct {
	// The (complete) local name that will be advertised. Optional, omitted if
	// this is a zero-length string.
	LocalName string

	// ServiceUUIDs are the services (16-bit or 128-bit) that are broadcast as
	// part of the advertisement packet, in data types such as "complete list of
	// 128-bit UUIDs".
	ServiceUUIDs []UUID

	// Interval in BLE-specific units. Create an interval by using NewDuration.
	Interval Duration

	// ManufacturerData stores Advertising Data.
	// Keys are the Manufacturer ID to associate with the data.
	ManufacturerData map[uint16]interface{}
}
```
there is only following elements coverd
```
0x01: Flags //though not configurable
0x09: Complete Local Name
0x03: Complete List of 16-bit Service Class UUIDs
0x07: Complete List of 128-bit Service Class UUIDs
0xFF: Manufacturer Specific Data
```

is this by design? or would you consider extending it?

nickajacks1, 2024-02-15:
Check out #123

dnlwgnd, 2024-02-15:
thanks for the pointer to #123. i briefly looked at it and it seems it does add the `Service Data` elements only on the receiving pieces of the code, i.e. in 
```
type AdvertisementOptions struct {
```
and
```
func (buf *rawAdvertisementPayload) addFromOptions(options AdvertisementOptions) (ok bool) {
```
`Service Data` was not implemented. 
How can we go forward on this? 
I could try to whip s.th. up but would require some guidance.

aykevl, 2024-02-17:
@dnlwgnd it appears that #123 was abandoned. Perhaps you can try to make a new PR based on that PR, including support for service data on the SoftDevice?

deadprogram, 2024-03-24:
Now closing since 0.9.0 has been released. Thank you!

PR #243 (2024-03-18): add ServiceData advertising element
this PR add support for ServiceData advertising elements (0x16: Service Data - 16-bit UUID)
It is based on the work of #123 of which I merged all commits.
fixes #241 

I added
 - handling of advertising (sd, linux)
 - handling of scan (sd, linux)
 - bugfixes
 - testcase

I tested successfully 
 - advertise with ServiceData from softdevice
 - scan with ServiceData from softdevice
 - scan with ServiceData from linux

There is an error I can not figure out when advertising with ServiceData on linux. I keep getting:
```
panic: failed to start adv: bluetooth: cou [... cut]
closed 2024-03-24 by deadprogram