tinygo-org/bluetooth #245: How to add custom service guide and characteristic uuid

nithin-gunamgari, 2024-02-29:
I connected a Bluetooth device which has vendor specific service and characteristic uuids. How do I add them in here so that I can read the notifications using /examples/heartrate-monitor.

aykevl, 2024-03-02:
There are two functions you can use to create a custom UUID:

* [`bluetooth.New16BitUUID`](https://pkg.go.dev/tinygo.org/x/bluetooth#New16BitUUID)
* [`bluetooth.NewUUID`](https://pkg.go.dev/tinygo.org/x/bluetooth#NewUUID)

For example, you can pass a UUID (consisting of 16 bytes) in an array to `NewUUID` to create a new UUID value.

nithin-gunamgari, 2024-03-04:
Thanks for the response. I did use NewUUID function and added a service and characteristic. But it says you can't edit the service_uuid or characteristic_uuid right. I want to use the code in heartrate-monitor for my own service and characteristic uuids. Can I you please help me with that?

aykevl, 2024-03-05:
You don't need to change service_uuid.go or characteristic_uuid.go. Just create your own UUIDs manually by calling `bluetooth.NewUUID`.
Here is how I've done it:
https://github.com/aykevl/things/blob/master/watch/ble.go#L141

But basically, just call `bluetooth.NewUUID([16]byte{ ... your uuid ... })`.

aykevl, 2024-03-05:
Closing, as this is not a bug.

closed 2024-03-05 by aykevl