tinygo-org/bluetooth #248: How to "DisableNotifications"

roffe, 2024-03-12:
I have a application that needs to be able to connect to the same device, disconnect from it and then re-connect again multiple times

It seems once enabled a notification keeps running and blocks the freeing of the device with Disconnect. Only solution i've found is to exit the application and restart it which is a very bad workflow,

Has anyone been able to solve this on Windows platform?

ansoni-san, 2024-03-14:
Yeah there's some changes needed for notifications on Windows to properly free everything on disconnect. 

We haven't needed notifications just yet but will need them in a few months. It shouldn't be too hard to make changes to this library. 

I'm sure a patch would be accepted if you make one. 

Otherwise we (at my project) probably will in a few months.

roffe, 2024-03-14:
> I'm sure a patch would be accepted if you make one.

I have no clue how any of this works, this is my first contact with BLE ever in my coding career so we just have to pray someone who knows how to helps

wwwfeng, 2025-04-24:
I got the same question,can't disconnet on windows. Who can fix it,Disconnect API doesn't work.

jagobagascon, 2025-07-29:
Labeling to close on next release, since this has been addressed by #368