tinygo-org/bluetooth #250: Warning, This project is not suitable for use in Windows!!

roffe, 2024-03-14:
There are several memory leaks in this software and things like enable notifications and disconnect does not work properly on windows

Hopefully this warning will help other people from falling in the trap beliving it's usable code for their project

https://github.com/tinygo-org/bluetooth/issues/248
https://github.com/tinygo-org/bluetooth/issues/130
https://github.com/saltosystems/winrt-go/issues/83

ansoni-san, 2024-03-15:
Hmmm, but this project is perfectly usable in commercial projects. 

I have several commercial products using it to send hundreds of Bluetooth requests a second to multiple devices on thousands of Windows machines.

If you have some issues, that are general issues, feel free to submit a patch 😊

This is how open source works. It may be the case that no one is currently using the way that you are using it.

deadprogram, 2024-03-21:
Closing, since no specific issue here.

closed 2024-03-21 by deadprogram