tinygo-org/bluetooth #251: Configuration of clockConfig = &clockConfigXtal causes crash on feather-nrf52840-sense

goderiich, 2024-03-14:
https://github.com/tinygo-org/bluetooth/blob/d0c7887b81e136df90b732d1162cdb335633aafa/adapter_nrf528xx.go#L41

My feather-nrf52840-sense was crashing using the advertising example and I was unable to nail down the issue as I wasn't receiving any panic messages and don't own a debugger. After hours of searching I came across this pull request comment: https://github.com/tinygo-org/tinygo/pull/4020#issuecomment-1938601968

By commenting out clockConfig = &clockConfigXtal I was able to get the bluetooth device broadcasting finally. Is this something that needs to be fixed in tinygo-bluetooth or does it need to be fixed in the tinygo machine package?

I have another feather-nrf52840-sense arriving soon and will test to ensure it's not a problem with my current board.

aykevl, 2024-03-16:
It sounds like this is not a bug in the Bluetooth package but rather here:
https://github.com/tinygo-org/tinygo/blob/6384ecace093df2d0b93915886954abfc4ecfe01/src/machine/board_feather-nrf52840-sense.go#L5

If it only works with `HasLowFrequencyCrystal` set to false, it sounds like it doesn't have such a crystal. (Which I find surprising, because it uses a BLE module that I believe has a crystal in other variants of the board).

> I have another feather-nrf52840-sense arriving soon and will test to ensure it's not a problem with my current board.

That would be a good idea. I have a nrf52832 dev kit for example (pca10040) and its crystal seems to have stopped working for me after a decent amount of abuse (traveling, throwing it in a box with lots of other boards, etc).

goderiich, 2024-03-17:
Thanks for your reply @aykevl I note that in the adafruit github comparing feather_nrf52840_express to feather_nrf52840_sense there does seem to be a difference

Express
```
#define USE_LFXO      // Board uses 32khz crystal for LF
// define USE_LFRC    // Board uses RC for LF
```

Sense
```
#define USE_LFRC    // Board uses RC for LF
```

My second sense will be arriving next week, so I'll test then raise an issue against https://github.com/tinygo-org/tinygo/blob/6384ecace093df2d0b93915886954abfc4ecfe01/src/machine/board_feather-nrf52840-sense.go#L5

closed 2024-03-17 by goderiich