tinygo-org/bluetooth #252: Memory leak on Linux with repeated device connections

jasday, 2024-03-19:
Expanding on the example `heartbeat-monitor`, I'm getting what appears to be a memory leak when re-connecting to a device after the first connection.

An example use case: 
- BLE Central starts adapter scan, awaits for device to advertise accepted service
- Device is found, central connects to device and enables notifications
- Central receives data, then disconnects device (at this point device could already have left range or been turned off)
- Upon device re-discovery, central connects again to device, and from this point onwards an increased memory usage is noted (increases by ~2mb every 10 seconds or so, with no hard limit - has reached at least 500mb).

Example code, edited slightly from the source:
```go
func (o *Bluetooth) Start() error {
        println("enabling")

	// Enable BLE interface.
	must("enable BLE stack", adapter.Enable())

	// Start scanning.
	println("scanning...")
        // Added a loop so we can start the scan again once we've finished with the current device
        for {
		ch := make(chan bluetooth.ScanResult, 1)

		// Start scanning.
		println("scanning...")
		err = adapter.Scan(func(adapter *bluetooth.Adapter, result bluetooth.ScanResult) {
			println("found device:", result.Address.String(), result.RSSI, result.LocalName())
                        // Using a health thermometer service as a test here, can be swapped out with anything else
			if result.AdvertisementPayload.HasServiceUUID(bluetooth.ServiceUUIDHealthThermometer) {
				adapter.StopScan()
				ch <- result
			}
		})
                var device *bluetooth.Device

		select {
		case result := <-ch:
			device, err := adapter.Connect(result.Address, bluetooth.ConnectionParams{})
			if err != nil {
				println(err.Error())
				return err
			}

			println("connected to ", result.Address.String())
			// Handle some logic here on the device
			println("Disconnected")
			device.Disconnect()
		}
	}

}
```

Go version: 1.22.1
tinygo/x/bluetooth: 0.8.0

Compiling onto Linux Arm64, with a IW416 SOC

Am I doing anything wrong based on the example code? The only real difference here is the added loop, which (hopefully) allows for scanning and connecting to different devices. (I need to be able to connect to multiple devices with various different services at unknown times, so this is just attempting to get the basics working, so I need the ability to restart the scan once the operation on the current device is finished)

Any ideas?

asteria-jacky, 2024-03-22:
I think the example code is good, instead there could be a problem in gap_linux.go. I see the number of goroutines keep incrasing together with memory usage, even after trying to connect once.

I can't see anywhere `(*device.Device1).UnwatchProperties` get called while there's only `(*device.Device1)WatchProperties`. This can make underlying dbus signal channel not get unregistered, then result in goroutine leak on every single deffered undelivered signal from dbus.
https://github.com/tinygo-org/bluetooth/blob/d0c7887b81e136df90b732d1162cdb335633aafa/gap_linux.go#L326C1-L331C3

I forked and added a `defer d.device.UnwatchProperties(d.propchanged)` and it seems okay so far.

They are trying to remove the dependency on muka/bluetooth, so I doubt this will be fixed.

jasday, 2024-04-02:
This seems to have been fixed in the latest v0.9.0 - When simply connecting to a device and disconnecting the memory no longer increases and will remain fairly stable. There is however increasing memory usage when connecting to a device and enabling notifications as mentioned in https://github.com/tinygo-org/bluetooth/issues/260.

deadprogram, 2025-01-02:
Thanks for the update on that @jasday 

Now closing this issue, since there is #260 to track new issue.

closed 2025-01-02 by deadprogram